KiCad/kicad-symbols #7: Transferring symbol data

SchrodingersGat, 2017-11-27:
I would like to start the transfer of symbol libraries from [kicad-library](https://github.com/kicad/kicad-library) to here. There are two major tasks to achieve:

## 1. Library Reorganization

The new libs will be sorted by function. Outline of examples for new library names - https://github.com/KiCad/kicad-library/issues/1402

This can be handled (mostly) by script. We just need to work out which symbols to transition.

My script for managing this can be found here - https://github.com/KiCad/kicad-library-utils/pull/184

## 2. Footprint Association

I am working on another script which will look through all the symbol files and work out which footprint associations need fixing. I'll have a PR up for that one soon.

Both of these steps can be performed for individual libraries, to make it easier. I will make a TODO file in this repo.

This means that once a library is marked as "complete" we can not accept any changes to it on the kicad-library repo

poeschlr, 2017-11-27:
> This means that once a library is marked as "complete" we can not accept any changes to it on the kicad-library repo

I would say once a library is in the process of being transferred. Or at the latest when to pull request is made. (Here it becomes a lot harder to tell people with open pull requests that their stuff will not be handled in the old repo.)

SchrodingersGat, 2017-11-27:
@poeschlr of course you are correct. It does not make sense to allow additions once the process is started.

antoniovazquezblanco, 2018-02-14:
Symbol transfer has been finished, am I right? Should this be closed?

poeschlr, 2018-04-19:
Yes i hope we did not miss something. If we did it will most likely be reported in a separate issue anyways.

closed 2018-04-19 by poeschlr