KiCad/kicad-symbols #19: Issues with audio lib

poeschlr, 2017-12-05:
- [ ] Some symbols have illegal footprint fields (not of the format [lib]:[footprint])
- [ ] Some have footprints assigned that do not seem to exist
- [ ] Off grid pins
- [ ] Some symbols have no entry in the dcm file

evanshultz, 2017-12-05:
@poeschlr Is this only for audio.lib? It seems the Travis output covers a lot of libs.

Furthermore, it appears Audio.lib contains loads of unrelated parts. This is not the library organization for audio symbols that was discussed at https://github.com/KiCad/kicad-library/issues/1402.

poeschlr, 2017-12-05:
This issue is only for audio.lib, but the travis error i linked comes from a pull request that added multiple libs.

I added the requirement to split the lib to the list of things to do.

sebastianbasierski, 2018-05-30:
RD5106A is an example of part with no longer valid datasheet link.
Also this parts availability is disputable.
What is the course of action in that case? Leaving bad link? Removing part at all?

evanshultz, 2018-05-31:
https://github.com/KiCad/kicad-symbols/tree/master/obsolete can preserve symbols that exist in the library but fall into this category.