KiCad/kicad-symbols #25: Issues regarding micro controller libs

poeschlr, 2017-12-05:
- [ ] Some have very long pin names. Remove alternative functions from them.
- [ ] Some footprint fields don't include the lib name

evanshultz, 2017-12-05:
From https://github.com/KiCad/kicad-symbols/pull/10#event-1372553510: Missing FPfilters, incorrect footprint fields, missing DCM entries (at least some of these are because the DCM still has the /), etc.

poeschlr, 2017-12-05:
Added to the list.

antoniovazquezblanco, 2018-09-18:
Every library has been checked with the library utils for rules S5.1 and S5.2 and the corresponding issues have been created in this repo. That should mark the last three bullet points out of this issue.

As for task number 1, should it be prebooked for v6?

evanshultz, 2018-09-18:
If the symbols aren't touched, but only named shortened, then schematics don't break. I vote for consistency of pin names if any changes are made by renaming pins to be uniform across all libraries, and then shrinking/moving/changing symbols only at a later date. But IMO doing anything now isn't bringing big value to the library for users as opposed to other librarian tasks. Just my feeling.

On this topic... It may not be worth the effort to do anything manually, though. I believe we should be able to extract all the symbol information from the existing libraries (we will assume for now it is all correct) and then generate YAML files for the library which would then be inputs to a script-based symbol generator. If we only do this work when we are ready/able to change symbols we shouldn't be wasting any effort.