KiCad/kicad-symbols #27: Issues with logic libs

evanshultz, 2017-12-05:
See https://github.com/KiCad/kicad-symbols/pull/11#issuecomment-349285725.

Items noted in that PR:
Here are some things that could be fixed.
- There are no packages or package types (just "SSOP" without pin count) listed in part descriptions
- No supply voltages given in descriptions
- Some hyphens in FPfilter, such as 74LVC1G02
- Some descriptions use all lowercase letters
- 74LS09 is describes as "Open Collect" instead of "Open Collector"
- The datasheet links in Logic_74xx don't work for me
- At least in the IEEE libs there are some lowercase letters for pins names; do we allow that?

There was a recent discussion about having a separate power symbol for even single-gate logic. I still support this so I want to resolve that discussion. I believe @jkriege2 was with me on this point.

jcyrax, 2017-12-28:
Hi

I can help with this issue if nobody has started working on them.

jcyrax, 2018-01-09:
@SchrodingersGat @jkriege2 @poeschlr I wan't to inform just in case that I started working on this problem so that nobody is doing it at the same time.

jkriege2, 2018-01-10:
Hi!

note that @bobc scripted some of these symbols: https://github.com/bobc/kicad-symgen and it might be a good idea to use/extend that generator!

Best,
JAN