KiCad/kicad-symbols #28: Issues with "Singles" libraries

evanshultz, 2017-12-05:
Sounds like a relationship problem...

See https://github.com/KiCad/kicad-symbols/pull/13#event-1372558892.

antoniovazquezblanco, 2018-09-18:
I can generate a report on S5.1 and S5.2 rules and split it by libs so that smaller and more concise issues can be generated. Those would be easyer to attack than this.

I am having troubles with the checklib script and some encoding messages. I will try to fix that and after that I will generate the report.

antoniovazquezblanco, 2018-09-18:
Peding libs:
- [ ] Driver_Motor.lib
- [ ] FPGA_Xilinx.lib
- [ ] FPGA_Xilinx_Artix7.lib
- [ ] FPGA_Xilinx_Kintex7.lib
- [ ] FPGA_Xilinx_Spartan6.lib
- [ ] FPGA_Xilinx_Virtex5.lib
- [ ] FPGA_Xilinx_Virtex6.lib
- [ ] FPGA_Xilinx_Virtex7.lib
- [ ] Interface.lib
- [ ] Interface_CAN_LIN.lib
- [ ] Interface_Ethernet.lib
- [ ] Interface_USB.lib
- [ ] LED.lib
- [ ] MCU_Intel.lib
- [ ] MCU_Microchip_PIC18.lib
- [ ] MCU_Microchip_SAML.lib
- [ ] MCU_Module.lib
- [ ] Memory_Flash.lib
- [ ] Memory_RAM.lib
- [ ] Memory_UniqueID.lib
- [ ] Power_Protection.lib
- [ ] Reference_Voltage.lib
- [ ] Regulator_Linear.lib
- [ ] Regulator_Switching.lib
- [ ] Sensor_Temperature.lib
- [ ] Timer.lib
- [ ] Transformer.lib
- [ ] Transistor_BJT.lib
- [ ] Transistor_FET.lib
- [ ] Triac_Thyristor.lib

antoniovazquezblanco, 2018-09-18:
All the problems included in this issue have been moved to smaller tasks. Closing this report in favor of the smaller units.

closed 2018-09-18 by antoniovazquezblanco

evanshultz, 2018-09-18:
@antoniovazquezblanco 
Just a question why make so many little issues when a checklist in this one big issue would work also? Is it because you prefer to have smaller pieces of work that can be closed rather than one massive task? Either way can work and I'm not chastising you, but it seems like a lot of effort to create so many issues so I just wanted to know your preference.

antoniovazquezblanco, 2018-09-19:
I prefer small issues that can be closed. Maybe this was a little bit too much but nevermind.

I believe the simpler and focused the issue is, the easier to close it is.

Ratfink, 2018-09-19:
This way we could assign each small issue individually to avoid duplicated work, like happened in some parts of #520.  Plus, all the tiny issues can be added to a project so we can easily track progress without having to look at comment diffs.