KiCad/kicad-symbols #29: Issues with DSP libraries

evanshultz, 2017-12-05:
See https://github.com/KiCad/kicad-symbols/pull/9#issuecomment-349467054: missing or wrong default footprints, broken FPfilters, etc.

SchrodingersGat, 2017-12-05:
The `DSP_` libraries require attention to fix the broken footprint associations

antoniovazquezblanco, 2017-12-11:
I use Microchip DSPs and the looks are not consistent between different symbols.

Also microchip, depending on the package changes pin numbering so I think it would be a good idea to provide a symbol for each packaging option otherwise users may find problems in the future but this requires a good amount of work I cannot contribute right now.

bobc, 2017-12-13:
DSP_Freescale contains only 1 component which is quite obsolete, not worth fixing, maybe worth moving to "old and obsolete" category. Also, all Freescale parts are now branded NXP (at time of writing !).

DSP_Texas also contains 1 component which is obsolete. Also suggest move to "legacy, not maintained" category.

I submitted PR to fix DSP_Microchip_DSPIC33. I did not change the existing packages there which seem to be correct for 44 pin QFN and TQFP.

aracosta1, 2020-08-05:
Hello, I can help by adding the Texas TMS320F28xxx series. 
Should the multiple functions of every GPIO be indicated or it's better to just call the pin with the only GPIO number ?

Since the devices have a lot of pin should we creata a single component or a multiple unit component to have smaller object in the schematics ?

chschlue, 2020-08-05:
Generally, we add two pin functions max.
Whether it's better to split symbols always depends, but if you add `_Split` to those it's easy to add a monolithic version later.