KiCad/kicad-symbols #36: RJxx symbols are oriented with their pins downwards instead of towards the right like other connector symbols

poeschlr, 2017-12-06:
See discussion in https://github.com/KiCad/kicad-symbols/pull/31

lorem-ipsum, 2017-12-06:
I've started a cleanup of all the registered jack / modular connectors in a [branch](https://github.com/lorem-ipsum/kicad-symbols/tree/rj-cleanup) of my fork. I will fix the orientation at the same time.

lorem-ipsum, 2017-12-06:
I have updated all connectors, but I am currently not sure what to do with the RJ45-TRANSFO. According to the old description it is based of a specific component from Wurth which I couldn't find with a quick search. In general the RJ45 with transformer is a generic component, but i don't think the pin numbering is the same for all connectors. Proper handling will most likely require manufacture specific components, which can hopefully be handled mostly by aliases.

[image]