KiCad/kicad-symbols #54: Problems with regulator libs

poeschlr, 2017-12-13:
- Wrong footprints assigned.
    - Converters_DCDC_ACDC has been split in Converter_DCDC and Converter_ACDC
    - multiple symbols reference `Package_TO_SOT_SMD:TSOT-23-6_MK06A`, this does not seem to exist under that name any more.

Electrical pin types for some of these symbols might be wrong.

More details see travis report of the PR that transferred these libs.

jkriege2, 2017-12-16:
The package is simply called `Package_TO_SOT_SMD:TSOT-23-6`

antoniovazquezblanco, 2018-09-18:
Convertes have already been fixed by my MRs a while ago AFAIK.

A list of detailed issues with footprints has been created in this repo (See #28).

For that reason I vote in favor of closing this issue.

myfreescalewebpage, 2019-03-18:
Agree with @antoniovazquezblanco too, closing.

closed 2019-03-18 by myfreescalewebpage