KiCad/kicad-symbols #56: Should smart fets (intedned for switching batteries) be placed under power_management or transistor_smart (inteligent)?

poeschlr, 2017-12-13:
Some intelligent FETs have been moved into power_management via pull request https://github.com/KiCad/kicad-symbols/pull/42. I merged it to get all the other contributions in there. But i feel the question about smart fets in general still stands.

Infineon (which hosts the datasheets for internal rectifier datasheets, might mean they own them.) places all their smart (intelligent) fets into the smart fet category. (No matter their originally intended application.)
https://www.infineon.com/cms/de/product/power/smart-low-side-high-side-switches/

[ITS428L2-DS](https://www.infineon.com/dgdl/Infineon-ITS428L2-DS-v01_01-en.pdf?fileId=db3a30431c69a49d011cdcc9a1fd41df) has the exact same features as [IPS6011](https://www.infineon.com/dgdl/ips6011pbf.pdf?fileId=5546d462533600a4015355c3db3d1625) (including reverse battery protection.) The later is one of these transistors placed into power_management.

poeschlr, 2017-12-13:
@SchrodingersGat, @jkriege2, @evanshultz what do you guys think?

jkriege2, 2017-12-13:
Hi!  from the Infineon page I see that they have them in `Power | Smart Low-Side & High-Side Switches` ... why not use that? To me that makes sense, so I would put them either into `power_management` (they are used to manage power-supplies), or into a new lib `power_switch` (maybe split by low-side/high-side).

I would not expect them in transistors-libs, as they are much more involved than a "simple" discrete transistor. Also they are not simply plugin-replacements for a transistor with all the intelligence hidden, so I would rate them their own lib.

Best,
JAN

poeschlr, 2017-12-13:
`power_switch` is definitely better than power_management. Maybe `power_switch_smart` to indicate that there is some intelligence build in.

SchrodingersGat, 2017-12-15:
Should power-switch also include things like ideal diode controllers? I would imagine so.

poeschlr, 2017-12-15:
I don't think ideal diode controllers are similar enough to such smart fets to be thrown into the same lib.
One is intended to replace transistors (plus the transistor driver), whereas the other is meant to replace diodes.

I am not even sure any more that switch is the best name for smart fets. True switches (even transistor switches) allow current flow in both directions. Whereas smart fets (like fets) are designed for the current flowing in one direction.

SchrodingersGat, 2017-12-15:
Digikey has such things under [Power distribution switches / load drivers](https://www.digikey.com/product-detail/en/infineon-technologies/AUIPS6011R/AUIPS6011R-ND/3455089)

I think it is a good idea to look at how someone like DigiKey organizes them because they categorize components from multiple manufacturers, whereas each manufacturer has their own scheme.

In a general sense, I think that the "purpose" of these devices is for power "distribution" - same goes for ideal diode controllers. 

Have a look at the various `PMIC_xxx` options here - https://www.digikey.com/products/en/integrated-circuits-ics/32

myfreescalewebpage, 2019-03-15:
@poeschlr it seems the parts have been added to power_management since you have opened this issue. Do you think now it can be closed ?
Joel

myfreescalewebpage, 2019-03-23:
@poeschlr ping ? Thanks.

poeschlr, 2019-03-23:
This is something we can not do anything about until we get to discussing potential changes for the version 6 library. I would keep it but mark it as version 6

myfreescalewebpage, 2019-03-24:
Done.