KiCad/kicad-symbols #61: Find libs for symbols in legacy/microchip

poeschlr, 2017-12-14:
The symbols for the following component categories need to find a good home.
- [ ] Serial to parallel converter
- [ ] I2c and SPI I/O expanders
- [ ] Clock divider
- [ ] A/D converter (I assume this should be moved to adc.lib)

SchrodingersGat, 2017-12-15:
What about `Interface_Expander` for I/O expander devices?

SchrodingersGat, 2017-12-15:
A/D will go to `Analog_ADC`

myfreescalewebpage, 2019-01-17:
@poeschlr 

I suggest:
- Serial to parallel converter to "Interface" or "Interface_Expansion" library
- I2c and SPI I/O expanders to "Interface_Expansion" library
- Clock divider to "Timer" library
- A/D Converter to "Analog_ADC" as indicated by @SchrodingersGat 

By the way, is there a list of concerned devices to move ?

Joel

poeschlr, 2019-01-17:
All symbols have been moved into the new repo. This means you only need to look what is in the legacy libs to know what might still need to be integrated in the new organization.

myfreescalewebpage, 2019-01-17:
Thanks for the response I suggest closing this issue in this case. What do you think ?

evanshultz, 2019-03-15:
This case appears to be about moving symbols in the old `microchip.lib` (at https://github.com/KiCad/kicad-library/tree/master/library). So I checked it out and found a symbols that are missing or still have issues in the new lib:
- HV513K7 / HV513WG: missing
- MCP2200* (8 symbols): remove temp rating and packaging to consolidate to 3 symbols
- MCP23008 / MCP23008_SSOP: missing
- MCP23016: missing
- MCP23S08 / MCP23S08_SSOP: missing
- MCP3021: missing
- SY89312V: missing
- The new lib's USB2514B_Bi should have the "_Bi" remove since that's the extended temp version

Of course we don't have to add all these symbols, but that breaks down the actions that could be taken from this issue. Or have I missed something?

myfreescalewebpage, 2019-03-16:
@evanshultz great job, that's not a lot of symbol but in this case it is probably better to keep the issue opened just as a reminder :)