KiCad/kicad-symbols #76: touch screen controllers

jkriege2, 2017-12-20:
Several touch screen interfaces are currentlx in `Driver_Display.lib`. 
I would put them into `Interface_HID`, as they do not SRIVE a screen, but interface the HID-component (HID=Human Input Device)

poeschlr, 2018-01-10:
I agree that they would fit better in a lib called interface_hid

@SchrodingersGat if you agree could you change the label form question to help wanted?

SchrodingersGat, 2018-01-10:
What about display drivers or modules that integrate touch controllers?

myfreescalewebpage, 2019-01-17:
@jkriege2 I have made the following list of concerned devices:
- ADS7843E
- ADS7843E-2K5 which is to be removed, it's the same as ADS7843E but with different selling
- ADS7843EG4 which is to be removed (?), seems the same as ADS7843E, ordering info in the datasheet don't indicate differences
- ADS7843IDBQRQ1
- XP20469QF
- XP2046TS

Are you agree with this list ?

@poeschlr We can book this issue for v6.

Joel

evanshultz, 2019-01-21:
I haven't seen Jan in a while so you may not get an answer.

There only needs to be a single ADS7843 symbol. All the suffixes are non-functional stuff like RoHS-compliance and temp rating.

I corrected the PNs of the latter two items on the list.

myfreescalewebpage, 2019-01-22:
> There only needs to be a single ADS7843 symbol. All the suffixes are non-functional stuff like RoHS-compliance and temp rating.

That's what I was thinking too.

@evanshultz are you agree we can book it for v6 ?

evanshultz, 2019-01-22:
I'm fine with that. Doesn't seem urgent and require action more quickly. Certainly Rene should be aware and approve, of course.

myfreescalewebpage, 2019-01-22:
Yes, that's to follow it. @poeschlr maybe your opinion on this ?