KiCad/kicad-symbols #94: Directory structure

SchrodingersGat, 2017-12-23:
There are now LOTS of libraries in a flat structure which makes them difficult to parse. Now that we are organizing by function, should we also split by directory (for large groups, at least). Each library must retain the prefix because the library name is used as the "nickname"

e.g.

* Sensor
   * Sensor_Pressure
   * Sensor_Temperature
   * Sensor_Motion
* MCU
   * MCU_Cypress
   * Microchip
      * MCU_Microchip_PIC10F

jkriege2, 2017-12-23:
Fine with me ... this should then als make transition to the 1-file-per-symbol approach with the new fileformat (somewhere in future ;-) easier ...

poeschlr, 2017-12-23:
The sym lib table does support this, so why not.

SchrodingersGat, 2017-12-26:
If we have time between completing the symbol transfer and release of v5, let's discuss then.

evanshultz, 2018-03-15:
@poeschlr 
This seems completed. Right? If so, would you mind closing this issue?

SchrodingersGat, 2018-03-16:
I'm happy to close this - perhaps can be implemented at a later stage but the library / community is not ready for another large sweeping change :)

closed 2018-03-16 by SchrodingersGat

poeschlr, 2018-03-16:
I think the lib for v5 could have handled it. But as i plan on also releasing a backport for v4 this might have created problems. So i agree that we might want to postpone this for v6.

Ratfink, 2018-03-16:
It doesn't seem like it should be hard to move files in to/out of directories and rename them in the process.  If you'd like, I could throw together a quick Python script to (un/)directory-ify the library for switching between v4 and v5 file layouts.  Then we might actually be able to do this change before v5.

Changing the file layout would have the little problem of breaking all the PRs we currently have open, though.  But that'll happen whenever we make this change, so it's not specifically an argument against doing it *now*.