KiCad/kicad-symbols #112: Footprint filters need to be adapted for the new footprint naming conventions

poeschlr, 2017-12-28:
I fear the footprint filters need to be adjusted for the new footprint naming convention. (Pitch -> P, ...)

SchrodingersGat, 2017-12-29:
A lot of such changes could be made with find-and-replace, i think.

poeschlr, 2017-12-29:
Yup atom would be powerful enough. I think we should do this either now and also require this for every further addition or we do it a few days before the release for all libs at once.

poeschlr, 2018-01-09:
Part of this might have been fixed with https://github.com/KiCad/kicad-symbols/pull/150

poeschlr, 2018-01-21:
@jkriege2 you where the author of #150. Did you fix everything mentioned here or is there still a problem?

jkriege2, 2018-01-21:
I think it's done (just checked with a short search in Notepad++)

closed 2018-01-21 by jkriege2