KiCad/kicad-symbols #133: RC1 Tagging

pointhi, 2018-01-05:
I got notification from the devs that tagging release candidates of the library would be helpful. Unfortunately I do not know when RC1 will be branched.

poeschlr, 2018-01-05:
We tagged all releases in the past and it is planned to tag them for the future as well.
As soon as we get information that rc1 is near we will tag the 3 repos.

SchrodingersGat, 2018-01-05:
There are a small number of remaining legacy symbol libraries that need to be transferred as a matter of priority before we are ready to tag :)

pointhi, 2018-01-05:
@SchrodingersGat 
and that stuff -> https://github.com/KiCad/kicad-templates

Ratfink, 2018-03-19:
Closing this, as each of the relevant repositories has a `5.0.0-rc1` tag.

closed 2018-03-19 by Ratfink