KiCad/kicad-symbols #164: Separate former linear library physically and logically

evanshultz, 2018-01-12:
I reworked the whole lib but separated symbols logically, so some footprints were not available for all ALIASes of a symbol. Each symbol should be unique both logically and also physically.

poeschlr, 2018-02-01:
This issue does confuse me. I really don't know what you mean by physically.

From another comment of yours (can't find it now) i would guess you want to make these parts more atomic. (Meaning not only separate the symbols from others but also the footprints)

myfreescalewebpage, 2019-03-18:
@evanshultz is this issue still valid today ?

evanshultz, 2019-03-18:
Yes

myfreescalewebpage, 2019-03-23:
Ok, but as indicated by Rene, can you explain more the purpose here, I'm also a little bit lost... Maybe with an example ?

evanshultz, 2019-03-23:
Maybe it's just the wording.

Logical = the symbol that contains the pins and their functions (which drives the netlist)
Physical = the tangible package that a part is delivered in; there can be many

Real parts have one logical representation but may come in many physical versions. For example, ON Semi sells LM393 in 3 physical variants: https://www.onsemi.com/pub/Collateral/LM393-D.PDF.

In the libraries above, they are mixed which can lead to user problems. I can't recall exactly where a problem was reported, but somebody (I think on the forum) had noted this after I cleaned up the old opamp library. To illustrate this, let's look at https://github.com/KiCad/kicad-symbols/blob/master/Amplifier_Operational.lib and investigate two symbols which don't have a default footprint specified so the package is chosen in CvPcb:
1. `ADA4807-1` has two footprint filters which are both valid packages for this symbol. See the last page of https://www.analog.com/media/en/technical-documentation/data-sheets/ADA4807-1_4807-2_4807-4.pdf. So even though there is no default footprint, the user should only be able to choose a footprint that the part can be bought in. We haven't specified the full MPN, but the user can't pick a logical and physical combination that doesn't exist.
2. `LM2902` is the first symbol that can truly be used generically. This logical symbol applies to many, many MPNs. The datasheet (http://www.ti.com/lit/ds/symlink/lm324-n.pdf) shows three packages available but there are more than those three footprints in the FP filter. Similar problems exist as you review all the ALIASes. It looks like as more ALIASes were added, footprints that applied to the new ALIAS were added but those footprints are _not_ available for all combinations KiCad allows a user to pick.

I finished, or nearly finished, doing this for the comparator library a while ago but didn't post it because I was sure there would be some discussion and it would break things which wasn't allowed then. I'm sure new comparator symbols have been added so it needs reviewed. I consider it a bug, but a minor one, so we will need to decide when to roll this out. I'll try to polish up and submit the comparator PR soon so it will give an example of what could happen, then we can decide exactly how to do this (in the best interest of KiCad users) and when to do it.

Let me know if that's still unclear or it makes sense what I'm talking about and why it could be a problem. Like I said, I think it's a minor bug.

myfreescalewebpage, 2019-03-24:
Thanks @evanshultz that's definitely more detailed, I agree with you on this subject. The best solution according to me is to have possibility to specify the footprint for each aliases of a symbol in KiCad, but this is maybe not planned yet...? I'm not aware at this moment of the v6 roadmap.