KiCad/kicad-symbols #165: Indicate on the schematic symbol the physical motion of a potentiometer

evanshultz, 2018-01-13:
See the previous discussion at https://github.com/KiCad/Potentiometers.pretty/pull/23.

Let me summarize the issue here. Let's use an audio volume control as the most basic and easy-to-understand circuit that surfaces the issue. In this case, pretend there is a rotary volume control which should make the music louder when turned CW and quieter when turned CCW.

Here is a circuit fragment:
[image]

The "CW" and "CCW" text allows the user to know how the wiper will move when the schematic is being constructed. The wiper must be at the top when the pot if fully CW so the signal nearly passes through the pot, and when the volume pot is turned fully CCW the wiper should be at ground so no signal is present at the second opamp.

Without these pieces of text, the schematic designer has no way to know where the wiper will be when the pot is fully at either end. The only way to sort it out would be to design the schematic, place the parts on the board to see ratsnest (routing is optional), and then trace out the connections while reading the pot datasheet. Things will work out half the time, and the other half of the time the user will need to flip the pot symbol so the ends of the element are reversed. It's truly a random 50-50 event.

By adding the CW and CCW text, and adhering to a standard in the library, we can guide the user to the physical behavior of the pot at the schematic level, allowing reasoning through the desired circuit operation and then properly connecting the pot terminals. No more guessing and going back to fix the schematic (netlist) for things the user could not have possibly known before layout.

There are other types of pots, too, besides rotary ones. Slide potentiometers are one example. In that case, instead of "CW" and "CCW" the pot pins would need a different name. But in the end the user will still need to know which way the wiper is moving as the slide pot is moved. Otherwise you're back to rewiring the schematic. Or, in this case, one could flip the slide pot footprints 180 degrees.

I hope this makes the problem more obvious. Does it help? Some nomenclature needs to be developed for the pot pins, then this needs to be codified so it's not broken in the future and KiCad users will have confidence in the pot symbol and footprints. Likely some footprints need to be updated as well.

bobc, 2018-01-13:
CW and CCW can be ambiguous, depending on how you look at it. I prefer to use an arrow which indicates increasing magnitude, which can be used for linear and rotary pots e.g:

https://www.jameco.com/Jameco/workshop/JamecoBuilds/schemekey5.jpg
[image]

myfreescalewebpage, 2019-03-18:
@evanshultz the above comment from @LarryJoy is helpful I think, what do you think of it ? We should also think to a new name for the CW potentiometers, but I can make a PR if we agree. Currently in te lib: "R_POT", "R_POT_Dual", "R_POT_Dual_Separate", "R_POT_TRIM", "R_POT_TRIM_US", "R_POT_US".
Joel

evanshultz, 2019-03-19:
All of the above comments support having some indication of pot travel. I wasn't sure how to handle non-rotary pots (CW and CCW don't work) and Bob suggested an arrow.

I'm OK with an arrow but the exact details aren't fully nailed down, such as which end is pin 1 (does the arrow point toward pin 1 or away from it?). Is anything else not known?

So a bit more definition and then implementation.

LarryJoy, 2019-03-21:
I am using a Bourns PV37W502C01B00 (12 turn) trimmer potentiometer. The schematic diagram of the pot shows terminals 1, 2, and 3 as follows: terminal 1 is one end, terminal 3 is the other end, and terminal 2 is the wiper. The CW (clockwise) direction is marked as being terminal 3. So if you are going to show an arrow for travel it should be from terminal 2, the wiper, with the arrow head pointing towards terminal 3.
--Larry
P.S. I had a very hard time getting to this posting. I went to GitHub directly, I have an account, and tried searching but nothing worked. The only way I got here was following the URL from the e-mail I received.

kenneth558, 2019-04-09:
No harm done, I was just referencing your comment in a closed thread where the Bourns PTV09A specifically was being discussed.  That part does not appear to use a generic footprint, but rather a specific one which is why I figured it could just as well conform to Bourns' numbering.  Sorry if I'm wrong, but I'd be a little surprised if it is using a symbol f/p since it is titled "Potentiometer_Bourns_PTV09A-1_Single_Vertical"