KiCad/kicad-symbols #174: How do generic components now work?

bpgerber, 2018-01-18:
I have been using kicad-symbols as my library now in addition to kicad-library.

It seems the generics are either not put in yet or using a different convention. I am trying to understand it.

Currently, the generics of inductors, resistors, caps, etc... are all under the pspice library (only non-CamelCase library) and the diode and capacitor at least are unusably large. Is it against the convention to put all passive components around the size of the resistor so common components like capacitors are actually usable with most IC schematic components?

Pictured is the difference.
[image]

jkriege2, 2018-01-18:
The lib has been transfered sometime before christmas, so it should be there, if you update:
https://github.com/KiCad/kicad-symbols/blob/master/Device.lib

poeschlr, 2018-01-18:
If you update the lib files, you still need to add them to kicad. For kicad version 4 you need to do this for every project via eeschema->preferences->component libs.
For kicad nightly (after sym-lib-table addition) you can add the libs to your global sym-lib-table.

bpgerber, 2018-01-18:
I have already added them to kiCAD. You are correct. It wasn't showing up in the filter as none of them have descriptions anymore.

poeschlr, 2018-01-18:
This happens if the lib is saved with a recent nightly build. It changes the library version information. This results in older kicad versions not seeing the dcm file content.

bpgerber, 2018-01-18:
So this will be fixed at the next update?

poeschlr, 2018-01-18:
I created a pull request to fix this particular problem. https://github.com/KiCad/kicad-symbols/pull/175

poeschlr, 2018-01-21:
We will create a separate branch for kicad 4 compatibility. See discussion in https://github.com/KiCad/kicad-symbols/issues/191. Commit https://github.com/KiCad/kicad-symbols/pull/175/commits/e60f1f1cee20973deb0177e32e98ceae93b3d5cb should be save to use until then.

closed 2018-01-21 by poeschlr