KiCad/kicad-symbols #183: Broken datasheets for NXP logic ICs

poeschlr, 2018-01-20:
Originally reported by @bobc in https://github.com/KiCad/kicad-library/issues/1802

---

NXP logic products are now transferred to Nexperia, so a lot of datasheet URLs have stopped working. :)
It should be possible to fix with a global search/replace.

herostrat, 2018-06-01:
Hey,

i found this information from a nexperia datasheet.

> Instead of http://www.nxp.com, http://www.philips.com/ or http://www.semiconductors.philips.com/, use http://www.nexperia.com

I can update the links in the libraries.
(If I find out how to fix the fact that I already made a PR of the masterbranch of my fork -.-)

evanshultz, 2018-06-01:
Be careful! This is not a global search and replace situation.

NXP sold off part of their business ("Standard Products") to form Nexperia a couple years ago. See http://media.nxp.com/phoenix.zhtml%3Fc%3D254228%26p%3Dirol-newsArticle%26ID%3D2177262.

Some former NXP parts ("Bipolar Power") are now branded WeEn. See http://media.nxp.com/phoenix.zhtml?c=254228&p=irol-newsArticle&id=2118078.

And NXP also retained some products.

Within some libraries (like 74xx) you may be able to search & replace, but not over the entire library.

myfreescalewebpage, 2019-01-16:
Fixed few days ago at https://github.com/KiCad/kicad-symbols/pull/1388. If more links are broken, suggest simply to list them in a new issue or create a PR. Closing this issue for now.

closed 2019-01-17 by myfreescalewebpage