KiCad/kicad-symbols #189: Symbols for spice simulation need different pin numbering than used for connecting to footprints

poeschlr, 2018-01-20:
Sadly there is (at the time of writing) no way to define different pin numbers for spice simulation and footprint connection. As ngspice seems to be inflexible with regard to pin numbering we might need to add specialiced symbols for simulation purposes.

Original discussion with more details: https://github.com/KiCad/kicad-library/issues/182

stambaughw, 2018-01-20:
> Original discussion with more details: KiCad/kicad-library#182

Rene,

There is no need to add special symbols for spice simulation.  To change
the spice node sequence to match the symbol pin out with the spice
model, create a custom field "Spice_Node_Sequence" with the correct node
sequence using integers separated with spaces.

Cheers,

Wayne

poeschlr, 2018-01-21:
@astroelectronica have a look at the information provided by @stambaughw. There is a symbol field that allows defining a different pin order for the spice export then used in the symbol.

astroelectronica, 2018-01-21:
@stambaughw With some model files that is not possible because it's encrypted. In other cases, model file have not all pins, i.e., a MOSFET in SOIC-8 package, with several drain or source pins. The model just shows one connection for that. 

Passives and semiconductors loss with a subcircuit (.subckt) associated. Circuit reference can't be C, L, Q, D, .... and all them turn to U.

I'm not sure if to modify simulation models is a good idea, because these files are verified by manufacturer.

Thanks.

stambaughw, 2018-01-21:
> @stambaughw <https://github.com/stambaughw> With some model files that
> is not possible because it's encrypted. In other cases, model file have
> not all pins, i.e., a MOSFET in SOIC-8 package, with several drain or
> source pins. The model just shows one connection for that.

Please do not include encrypted spice models with kicad symbols.  This
violates every ethical argument for using open source.  If users want to
use them for your own projects, that is fine but we should not be
supporting them in our symbol libraries.

> 
> Passives and semiconductors loss with a subcircuit (.subckt) associated.
> Circuit reference can't be C, L, Q, D, .... and all them turn to U.

This behavior is defined by the spice standard.  Any .subckt statement
requires a U reference.

> 
> I'm not sure if to modify simulation models is a good idea, because
> these files are verified by manufacturer.

We should not modify the symbol file, we should use the node sequence
file to map the symbol to the spice model.  There may be some odd models
with control inputs that require custom symbols but most of the spice
models I use do not require custom symbols.

evanshultz, 2019-02-08:
@poeschlr 
Can this be closed? If so, please close it.

poeschlr, 2019-02-08:
I think so.

closed 2019-02-08 by poeschlr