KiCad/kicad-symbols #190: KLC-modifications/additions: OPAMPS/logic gates and power-units

jkriege2, 2018-01-20:
transfer of old issue https://github.com/KiCad/kicad-library/issues/1839

# original post:

I would like to suggest two additions/modifications to KLC, in part inspired by the discussion https://github.com/KiCad/kicad-library/pull/1828:
1. On S3.8 (http://kicad-pcb.org/libraries/klc/S3.8/). For logic and OPAMPs that's a bit inconsistent. I would suggest to ALSO separate the Power-ports for two special cases, namely logic and OPAMPs (I think these are also the most common devices that come as multi-unit-symbols). My rationale would be: If I design a big logic-circuit, it's good to collect all the power-units at one location and connect them to the appropriate power-plabes (+caps and what not). This tidies up the schematic significantly. If we discourage extra power-symbols for single-unit logic-symbols, this will lead to a mixture, as some logic ICs have their power-pins inside the schematic (may be annoying if you change  power-suuply to them, as you have to do it for every device separately) and some are collected somewhere. Also this would allow us to make unify the single-gate-logic families. Maybe (maybe!) it is possible to design the power-symbols in a way that they can be positioned over the actual symbol (at least for gates), so they can (if wanted) be "combined" to a single device. I think @bobc has some examples of that as alternative symbol style.
2. In addition I would like to see the same exception for OPAMPs for the same reason and with the same style requirement of beeing combineable (I think that ios already common for many multi-unit OPAMPs.
[image]
3. _[item 3 removed from old post as it does not match thematically and was not generally agreed on]_

------

# Discussion

I think this was mainly agreed on in the old issue. So if nonone has any additions/problems with that I will propose some wording soon. Please leave your comments here, so they can be incorporated in the final wording.

bobc, 2018-01-20:
I'm not sure what exactly is being proposed here. Overlaying the power pins looks rather ugly, and only works for certain symbols like AND.

Power units were introduced to fix a specific problem which had no other reasonable solution, and are still not ideal. I'm not convinced we need to change the graphical style again, there seems to be very minimal gain. I think we need to be careful whether we are fixing real problems, or tinkering with the style for the sake of it. 

The huge problem here is that KiCad does not allow per-unit placement of ref/value fields, which is very restrictive on how units can be designed. Additionally, ERC does not check the power unit is placed, so we should avoid creating unnecessary power units.

I suggest until those restrictions are removed, we keep the current rules.

jkriege2, 2018-01-20:
I designed these rules especially for OPAMPs (for single-gates thats more an extension), as there it is often convenient to have the power-pins near one of the say 2 OPAMPs in a schematic.

Generally I think this can be broken down to two requirements:
1. make power-symbols with no box
2. if you do: make pin-spacing so that it fits with the OPAMPs/logic gates. 

I think we need/should have rules for both and a defined style anyways, otherwise you'll see incompatible power-symbols that have different pin-spacing. So while requesting that we can make them in a way, people can use them this way if they choose so and any other way else.

Also this way: As soon as there is a way in KiCAD to designate one unit as power-unit, we can just modify the existing symbols and do not need any other style changes. the same for label-placement. Also here I think the proposed solution leaves all possible usages open and as soon as we have the required features, we only need to modify the existing symbols a bit and do not need to overhaul all of them (which we've seen in the past is quite tedious work).

bobc, 2018-01-20:
Ok, you seem to have conveniently ignored my suggestions.

I don't mind a rule that is specific to op-amps, but what you suggest simply doesn't work for logic gates. That will give me no option but to ignore the KLC rules...

jkriege2, 2018-01-20:
I don't really see your problem: I just propose to make the power-symbol in a way that ALLOWS for moving it over the smal-gate or OPAMP-symbol. If you don't want to place it there, simply don't in your scematic. In the end this will only impose a style-requirement for the symbol (e.g. pins Xmil apart, lines pointing inwart, specific line-length, no box). In the very least this will help to unify the style of power-symbols and eases their combination across several parts.

Of course this will not be usable for "large" things like FPGAs or CPUs with possibly a lot of different power-signals.

I would like to see that also for logic gates, as it would set then a style also for single-gate logic, which allows using them in larger logic assemblies where power-symbols might be colllected in one spot and for small schematics where it might be more convenient to have symbol and power-symbol close to each other.

bobc, 2018-01-20:
Ok, forget it, do your way. I give up.

jkriege2, 2018-01-20:
I'm not sure what you wuld propose ... and why there is a problem in proposing a style for the power-symbols?

evanshultz, 2018-01-22:
I still like this and agree. My comments on the original post are still current.

evanshultz, 2018-03-13:
@poeschlr @jkriege2 @Ratfink @Misca1234 
Any new input? Where do we stand on this? There hasn't been any action here in a while but it would be nice to settle on a style here so we have direction for symbol contributions.

Ratfink, 2018-03-13:
If y'all want to do it that way, go ahead, but I'm with bobc on this one.  It seems to me like there isn't much to be gained by this change, and changing it creates extra busy work for us.

evanshultz, 2018-05-18:
Strongly related but not totally the same topic...

See https://github.com/KiCad/kicad-symbols/pull/578. Opamps and other opamp-like symbols now have (or should have tomorrow) a separate unit for the power gate if there are multiple logical symbols. Single opamps are still one symbol. The primary motivator for this was that the power pins previously were on every gate, which broke CvPcb's pin count filter and potentially allowed bad net shorts.

That means the per-gate text placement and lack of alerts for unplaced power pin units will be more severe.

One outstanding issue with doing that is when there are multiple power pins (like positive, negative, and ground). This means the power unit needs 3+ pins and it may not be clear what those pins are. One option would be to put the triangle gate on the power unit. That still allows overlaying the power unit but it avoids floating pins:
[image]

We could also consider smaller pin name text, as discussed and seemingly accepted at https://github.com/KiCad/kicad-library/pull/1611, which could allow more clarity.

jkriege2, 2018-06-30:
I better like having them floating. For me the triangle also indicates OPAMP-functionality, but all the necessary pins then are not there in the power symbol. Also it makes the power-symbol unneccessarily large, when placed separately from the actual OPAMP-part. FInally I think floating is kind of standard in other software?

poeschlr, 2018-08-30:
Am i right in assuming this has been fixed?

evanshultz, 2018-08-30:
Perhaps, but this issue is also covering single-gate symbols. I recall discussions about single-gate symbols having a separate power gate many times. Currently only multi-gate symbols have a separate power unit. If you will only allow single-gate symbols to have just a single unit, then I think this can be closed. (I like separate power pins but I understand there's not only preference but limitations in KiCad that make this difficult right now.)

Best to get feedback from Jan, though, in case there's something still open.

poeschlr, 2018-08-30:
Any further changes would need to wait till we are allowed to break existing designs again (so when we start with v6 development.)
So we either mark this as "prebooked for v6" or close it if everything is done.