KiCad/kicad-symbols #213: KLC: Better define "center of symbol"

matthijskooijman, 2018-01-23:
The [KLC S3.1](http://kicad-pcb.org/libraries/klc/S3.1/) says "the symbol must be centered around the origin (0, 0) in the symbol editor", but it is not explicit about what the center means exactly. Does not mean the center of the symbol body? Or also include the pins? Does it also include the refdes and other labels? Last time I checked the checker tool uses the pins only, but that is known to be insufficient (and completely breaks for symbols that have pins on only one side).

bobc, 2018-01-24:
There was a discussion about this before, and no one could define what the center should be, or how to make a comprehensive rule.

I would include pins and graphical symbols as part of the calculation, but exclude fields (the position can vary on the schematic).

This is one of those rules that seem to have no purpose other than "OCD/aesthetic consistency", there is no real reason why the symbol needs to be centred. When the symbol is placed on the schematic, it's hard to even tell where the origin is, and doesn't matter to the user. Given that, I think we can afford to be quite fuzzy on the acceptable center.

Rules are great, but false positives create too much noise and really should be avoided.

poeschlr, 2018-01-24:
I would say the center should be somewhere near the graphical center because when rotating or mirroring the symbol in eeschema the user will expect that to be the anchor. This is already hard to explain. Now one needs to find a way to teach travis what to do.

bobc, 2018-01-24:
> I would say the center should be somewhere near the graphical center because when rotating or mirroring the symbol in eeschema the user will expect that to be the anchor. 

That does make sense, but it might be difficult to create a comprehensive rule that can be programmatically checked, if it is based on a subjective thing like "what the user expects" (I find not all users think the same way :) )

There are many symbols in Connector_generic that fail the KLC script, and I believe those were recently generated, so...

We could say "center of the graphical symbols, or if none then the center of the pins", where center is defined as center of the bounding box.

That would still likely have some exceptions which need checking visually, and is also non-trivial to calculate for arc segments. The KLC script also doesn't seem to take account of separate units, but globs them all together.

antoniovazquezblanco, 2018-01-24:
In my opinion the easiest way to go would be to just center based on the body of the component. When I rotate an element I expect it to pivot around the body (even if it only has pins on one side). I would create the axis aligned bounding box of the graphical elements that compose the body and would attempt to center that.

Anyways, I am sure there are many counterexamples in the libraries at the moment but this may silence many warnings of the checker script.

matthijskooijman, 2018-01-24:
@antoniovazquezblanco, that sounds like a good approach to me. For symbols with pins on both sides, these are usually symmetrical around the body, so centering on the body will also center on the pins, and mirroring and rotating keeps the pins in the same places.

For symbols with pins on one side, I would expect that rotating and mirroring keeps the body in the same place, so centering on the body also solves that.

I can imagine the actual rule being as you propose, but explicitly stating that a different center could be chosen for complex (graphical) symbols, and also state that pin grid alignment is essential and after that, rotation/mirror behaviour is something to keep into account?

Also, I just realized that the off-center exception to preserve pin alignment is actually slightly annoying, since it means that mirroring and rotating now moves pins around, so the component must be moved after rotating to keep the pins in the same place. I guess there is no fix for this, other than change the library format to include a separate "transform origin" and a "grid origin" (or change KiCad to do placement of a component on the grid based on a pin instead of the origin).

poeschlr, 2018-01-24:
@bobc Connector symbols with even number of pins create a problem as the center requirement contradicts with the pin on grid requirement. When we created the scripts we decided to give the pin on grid requirement a higher priority then the rule for centering the component.

This could be solved by allowing some deviation from the calculated center point. (+/-50mil in each direction might be enough.)

bobc, 2018-01-24:
I'm referring to connector symbols where the pins are on grid, but the center is on the body. KLC script flags an error:

```
Checking symbol 'Conn_01x01_Male':
  Violating S3.1
    Origin is be centered on the middle of the symbol
    Symbol not centered on origin
     - Center calculated @ (200, 0)
```

So currently I think the KLC script is checking the wrong thing, if what we actually want is to center the body.

matthijskooijman, 2018-01-24:
See https://github.com/KiCad/kicad-library-utils/issues/195 for an issue I previously reported about the checker scripts. The current script is indeed simple (taking into account only the pin positions, not the body, and not allowing the needed +/- 50mil off-center-ness yet).

jkriege2, 2018-01-24:
Please note that two improvements to the scripts have been merged recently:
1. https://github.com/KiCad/kicad-library-utils/pull/197 tries to identify the symbol box and calculates the center of that instead of the pins. This is especially valuable for e.g. 3-pin voltage regulators, where pin-ordering may be unbalanced
2. https://github.com/KiCad/kicad-library-utils/pull/200 only warns if the shift from the calculated center is larger than 50mil. below that it just prints out a text message!

bobc, 2018-01-24:
Nice! 

https://github.com/KiCad/kicad-library-utils/pull/197 will cover a lot of simple cases. It needs extending to handle multiple units, poly/arc etc, but it's a good start.

antoniovazquezblanco, 2018-07-16:
1. According to what has been done in the test scripts, if I understood correctly, http://kicad-pcb.org/libraries/klc/S3.1/ should be updated to state that only the body should be centered (not the pins). Is this correct?

2. Also scripts should be improved to deal with multi-unit schematic components. Am I right?

poeschlr, 2018-07-17:
Yes the script should deal with multi unit schematics.

I am not sure how to properly define the centering stuff. I would say symbols are ok if they look as centered as possible but this is not really something that we can use as the basis for the script.

antoniovazquezblanco, 2018-07-18:
I've moved the pending tasks into smaller units in the "corresponding place". I believe this issue is no longer needed.

I hope this helps with issue cleanup and organization.

poeschlr, 2018-07-18:
Thanks for your great work lately.

closed 2018-07-18 by poeschlr

matthijskooijman, 2018-07-18:
> Also, I just realized that the off-center exception to preserve pin alignment is actually slightly annoying, since it means that mirroring and rotating now moves pins around, so the component must [...]

I still wonder if this is something to consider. I'd create a separate issue for that too, but I'm not sure where that should live. Any suggestions?