KiCad/kicad-symbols #214: KLC: Text size requirements seem confusing

matthijskooijman, 2018-01-23:
The [KLC S3.2](http://kicad-pcb.org/libraries/klc/S3.2/) says:

 1. Text size must not be greater than 50mils (1.27mm)
 2. Text size must not be less than 20mils (0.508mm)
 3. All text fields must use text size of 50mil
     -  Pin names
     -  Pin numbers
     -  Value field
     -  Reference field
     -  Footprint field
     -  Datasheet field
 4. Pin numbers must use text size of 50mils
 5. Pin names should use text size of 50mils
 6. Smaller text size may be allowed (only for pin names) when required for small symbols or those with special geometry

This confuses me because:
 - 4. and 5. seem redundently restating what 3. already said.
 - 1. and 2. seem irrelevant (allowing text bigger and smaller than 50mil), since 3. states only 50 mil is allowed. The minimum size might be relevant for 6., but then it should probably just be added to 6. instead?
 - Shouldn't 6. also allow smaller size for pin numbers instead of pin names? I would expect small symbols to have problems with big pin numbers as well?
 - Is the list of fields in 3. exhaustive? It says "all text fields", which could include any extra (non-standard) text not shown in that list, but the list suggests that these are not subject of this requirement.

bobc, 2018-01-23:
Is your confusion theoretical, or do you have real examples that don't fit the rules? It's a lot easier to understand the issues if there is something concrete to look at.

poeschlr, 2018-01-23:
This rule could be written better. But i think the intention should be clear to everyone.

I think the following wording might be better:

---

1. All text fields should use a text size of 50mil.
2. Pin names and pin numbers can use a smaller text size. If this is required for small symbols or those with special geometry.
    -  The size is allowed to be between 20mil and 50mil in such cases.

This rule applies to the following Fields:
- Pin names
- Pin numbers
- Value field
- Reference field
- Footprint field
- Datasheet field

matthijskooijman, 2018-01-23:
@bobc, I noticed this while looking through the KLC for something else, so no specific components involved.

@poeschlr, your proposal looks good and adresses all my confusion :-)

evanshultz, 2018-01-23:
@matthijskooijman Thanks for pointing out that this could be improved. I think more rules were added, with good reason, but without going back and updating the entire clause.
@poeschlr Just a tiny bit of re-wording as a suggestion, if you don't mind.

1. All text fields (pin name, pin number, value, reference, footprint, datasheet) should use a text size of 50mil.
2. Pin names and pin numbers can use a text size as small as 20mil if the symbol is small or has special geometry.

poeschlr, 2018-01-25:
The new rules have been merged

closed 2018-01-25 by poeschlr

bobc, 2018-01-25:
Just a minor point, the date on recent changes should be 2018 :) http://kicad-pcb.org/libraries/klc/history/

poeschlr, 2018-01-25:
Thanks, pull request with the fix is commited: https://github.com/KiCad/kicad-website/pull/268

matthijskooijman, 2018-01-26:
Looks good, thanks!