KiCad/kicad-symbols #215: KLC: Guidelines for connectors

matthijskooijman, 2018-01-23:
This is a followup of https://github.com/KiCad/kicad-library/issues/1627 about connector symbols. Here are the things I think would need to be clarified about connector symbols in the KLC:

 - A general recommendation of how to make connector symbols look (rectangle with filled background, or a graphical representation with or without filled background). It seems the preference here is for a plain rectangle, but the current KLC contains an image of correct pin placement that contains a lot of these graphical connectors, which can be seen as a recommendation for them. Even if there is no strong preference for either, it might be good to mention both options and say both are acceptable?
- For rectangular connector symbols, it is customary to use small rectangles on the inside of the pins, but the KLC does not mention this. From the current symbols, only the generic connectors, USB and Atmel connectors have these rectangles. What's the rationale for using these rectangles? I just noticed that the USB symbol (see above) does not have this box on the "Shield" pin, is this because it does not correspond to a regular pin, but a mechanical mount? Or is it just an oversight?
 - For these small rectangles, [as I mentioned here](https://github.com/KiCad/kicad-library/pull/1736#issuecomment-337529247) there are various sizes in use (haven't checked the latest symbols). Some consistency here might be nice? The use of these rectangles also needs a bigger pin name offset than the default 20mil.
- #36 says connectors should have their pins to the right, but this is not stated in the KLC.

poeschlr, 2018-01-23:
We generally do not include detailed style guides in the KLC for each symbol type. This would bloat the KLC too much.

If i remember correctly there is some recommendation somewhere that the general style of other similar symbols should be followed.

LarryJoy, 2018-09-04:
Hello symbol library developers. Concerning the graphic symbols to be used for connectors. The "standard" that I am aware of and use is from IEC 60617, which appear in IEEE Std 315A (supplement to IEEE Std 315 that implements or presents the newer IEC symbols). The symbols do not use boxes or rectangles. What the symbols use is the dashed line to indicate mechanical linkage, that the terminals, whether male pins or female sockets, are connected together.

[image]

[image]

If these screenshots do not come through properly I will have to try something else. I try "dragging & dropping" and "selecting them" but all that appears is text. I use Ubuntu (18.04 LTS) so as far as I know there is no "pasting from the clipboard".