KiCad/kicad-symbols #217: Valve/tube schematic symbols

evanshultz, 2018-01-23:
See https://github.com/KiCad/kicad-library/issues/1580. It seems like a worthwhile conversation to complete so any direction on updating the lib is known.

bobc, 2018-01-24:
The question I always wonder with this type of thing is, are there any issues that need addressing, or is it mainly busy work? It tends to invite bikeshedding...

Apart from some off grid pins, I don't see much in the valve lib that actually needs updating. And given the legacy and niche usage, there doesn't seem much point spending time on it.

evanshultz, 2018-01-25:
@bobc 
If someone wants to contribute, then they can. Nobody is asking you to do it.

@Misca1234
Are you still interested in this?

bobc, 2018-01-25:
You have added the tag "help wanted".  Could you say exactly what help is needed? This would help anyone thinking of contributing to understand what is required.

Ratfink, 2018-04-18:
I don't think making valves look any way other than the normal IEC style would help anyone read a schematic.  So I'm opposed to the proposed changes in the old PR.

myfreescalewebpage, 2019-03-18:
@evanshultz @Misca1234 we have KLC S3.3 (http://kicad-pcb.org/libraries/klc/S3.3/) indicate background should be filled for this kind of symbol I think. I suggest this is the only modification to do on valves at this moment. What is your opinion ? I haven't checked the library to know how many symbol are concerned, but I can make a PR in this direction if you agree. If I see any major KLC issue I can also fix them (NC pins, pins length etc).
Joel

evanshultz, 2019-03-18:
By clause 2 of the rule you attached I'm not sure valves should be filled with background color. I think they are discrete devices.

Symbol centering isn't a problem, but the check script finds a number of things that should be fixed:
```
Checking symbol 'CK6418':
  Violating S3.1
    Origin is centered on the middle of the symbol
    > Symbol slightly off-center
    >   Center calculated @ (0.0, 50.0)
  Violating S3.6
    Pin name position offset
    Pin offset ouside allowed range
     - Pin offset (0) should not be below 20mils
  Violating S6.3
    Part meta-data is filled in as appropriate
    ALIAS CK548DX has metadata errors:
     - Missing all fields on 'Properties > Description' tab
Checking symbol 'EABC80':
  Violating S3.1
    Origin is centered on the middle of the symbol
    Symbol not centered on origin
     - Center calculated @ (-100.0, 0.0)
  Violating S3.6
    Pin name position offset
    Pin offset ouside allowed range
     - Pin offset (0) should not be below 20mils
  Violating S4.1
    Pin requirements
    Pins not located on 100mil (=2.54mm) grid:
     - Pin F1 (4), @ (-100, -350)mil
     - Pin F2 (5), @ (100, -350)mil
Checking symbol 'EC92':
  Violating S3.1
    Origin is centered on the middle of the symbol
    Symbol not centered on origin
     - Center calculated @ (-100.0, 0.0)
  Violating S3.6
    Pin name position offset
    Pin offset ouside allowed range
     - Pin offset (0) should not be below 20mils
  Violating S4.1
    Pin requirements
    Pins not located on 100mil (=2.54mm) grid:
     - Pin F1 (3), @ (-100, -350)mil
     - Pin F2 (4), @ (100, -350)mil
Checking symbol 'ECC81':
  Violating S3.1
    Origin is centered on the middle of the symbol
    Symbol not centered on origin
     - Center calculated @ (-100.0, -25.0)
  Violating S3.6
    Pin name position offset
    Pin offset ouside allowed range
     - Pin offset (0) should not be below 20mils
  Violating S4.1
    Pin requirements
    Pins not located on 100mil (=2.54mm) grid:
     - Pin F1 (4), @ (-100, -450)mil
     - Pin F1 (5), @ (100, -450)mil
     - Pin F2 (9), @ (0, -450)mil
Checking symbol 'ECC88':
  Violating S3.1
    Origin is centered on the middle of the symbol
    Symbol not centered on origin
     - Center calculated @ (-100.0, -25.0)
  Violating S3.6
    Pin name position offset
    Pin offset ouside allowed range
     - Pin offset (0) should not be below 20mils
  Violating S4.1
[... 124 lines cut ...]
  Violating S3.1
    Origin is centered on the middle of the symbol
    > Symbol slightly off-center
    >   Center calculated @ (0.0, 25.0)
  Violating S3.6
    Pin name position offset
    Pin offset ouside allowed range
     - Pin offset (0) should not be below 20mils
  Violating S4.1
    Pin requirements
    Pins not located on 100mil (=2.54mm) grid:
     - Pin K (2), @ (-100, -350)mil
     - Pin L (6), @ (0, 450)mil
Checking symbol 'STABI':
  Violating S3.6
    Pin name position offset
    Pin offset ouside allowed range
     - Pin offset (0) should not be below 20mils
  Violating S6.3
    Part meta-data is filled in as appropriate
    Component STABI has metadata errors:
     - Missing all fields on 'Properties > Description' tab
```

And if Stefan wants to propose any updates I'm fine with that.

myfreescalewebpage, 2019-03-18:
@evanshultz 

> I think they are discrete devices.

I'm not against this position, in this case ok for me to not fill the background.
I will have a look to the valve and try to fix the KLC errors. I think that many grid error will be difficult to solve because of the very specific arrangement of the pins on some symbols...

myfreescalewebpage, 2019-03-18:
@evanshultz I have just submitted this PR: #1649

Misca1234, 2019-04-06:
I left a comment in your new thread, mainly addressing the waiting push
https://github.com/KiCad/kicad-footprints/pull/1139