KiCad/kicad-symbols #228: Missing alias in connectors library

aurabindo, 2018-01-27:
Commit 2135c6433d4ecec8648f34c6ead40e3e19776e98 gives me a warning when opening my schematic:

```
15:36:34: Alias 'Conn_2Rows-67Pins' not found in library:

'/home/aj/kicad_sources/kicad_github/kicad-symbols/Connector_Generic.dcm'

at line 2037 offset 23
```

poeschlr, 2018-01-27:
Is it possible that your copy is corrupted? I just tested the lib and i don't get a warning.
I also looked at the files and the mentioned dcm entry also exists in the corresponding lib file.

bobc, 2018-01-27:
There appears to be mismatch in names? Conn_2Rows_67 vs Conn_2Rows-67Pins

poeschlr, 2018-01-27:
Yes i opened the wrong project with the wrong set of libs activated. (My bad)
Fix will be committed shortly.

jkriege2, 2018-01-27:
has been merged

aurabindo, 2018-01-27:
I can confirm, closing the issue. Thanks for taking care of it pretty quickly

closed 2018-01-27 by aurabindo