KiCad/kicad-symbols #232: SPI interface pins of a lot of symbols have the wrong electrical type.

poeschlr, 2018-01-28:
The following is only valid for symbols of devices that do not use the same pins for other functions.

For master devices the electrical types should be as follows:
- CLK = output
- Chip select (CS or ~CS) = output
- MOSI = output
- MISO = input

For slave devices:
- CLK = input
- Chip select (CS or ~CS) = input
- MOSI = input
- MISO = tri-state (Currently a lot of devices have the MISO pin as output. This would result in erroneous ERC messages if more than one slave is connected to the same master interface.)

jkriege2, 2018-01-28:
Two comments on that:
1. When we correct, we should take care to check for the existence of a CS-pin ... if it's not there it might well be that MISO is a true output
2. I checked the SDA/SCL pins of  I2C and made a PR to fix a few bugs: https://github.com/KiCad/kicad-symbols/pull/233

bobc, 2018-01-28:
I have a PR in progress for Memory_EEPROM (#222), I will it update for MISO pin types.

I would be a bit wary of doing global edits without checking datasheets, it could lead to more errors then we solve like that (well, I already made that mistake once!).

In all the libs I have reworked so far, they all have parts with pin types that are just completely wrong (e.g. power pins set to power out instead of power in, outputs set as inputs etc). I guess it might help to do some global edits, but most of the bad parts need a complete overhaul anyway.

jkriege2, 2018-01-28:
Yes ... For the I2C it was rather quick (with a text-editor global search you quickly find the ones with another type than the expected one and can check them ine-by-one then ...), but it's going to be checking each one separately in the end ... I fear