KiCad/kicad-symbols #237: Naming for the multi-footprint-approach

jkriege2, 2018-01-28:
Hi @evanshultz @poeschlr @pointhi @SchrodingersGat!

I'm currently reviewing @bobc's PR on EEPROM/EPROM: https://github.com/KiCad/kicad-symbols/pull/222 . There we have a lot of devices that are available in a plethora of 8-pin-packages without EP (e.g. DIP-8/SOIC-8/TSSOP-8) + a few with EP (e.g. DFN-8). How should we handle these cases in naming?
1. Have a specific symbol for each package.
2. Have a symbol for each pinout/pin-count, i.e. DFN-8 specific and all 8-pin-counts n one without footprint-field filled, but with FPFilters for all matching packages.

If option 2 is allowed, how do we handle naming? e.g. `25LC010A-MC` for DFN, but what to use for the multi-symbol, which does not coper all packages? Still use `25LC010A`?

Best,
JAN

evanshultz, 2018-01-29:
This is a bit of an aside, but when I reworked the old linear.lib everyone wanted the method where the FPfilter only included valid footprints for the symbol. So the symbols were separated logically and physically. I do not think your question above is this topic.

I like the first approach because it gives smaller file sizes and it's easier (at least for me) to identify the symbol you want when adding ALIASes later. I think that dropping the part of the MPN that identifies the package, as you've done above, is the way to go.

poeschlr, 2018-01-29:
I vote for one symbol per package. Until the new file format allows inheritance of symbol data this is the best approach as it allows setting of default footprints. (Default footprints reduce the risk that the user selects a wrong footprint.)

bobc, 2018-01-30:
The KLC http://kicad-pcb.org/libraries/klc/G2.1/ talks about defining generic and atomic parts. It gives some examples, but does not really express a preference for either style or any explicit guidance on which should be used. Therefore it gives the impression that either style is acceptable for submissions.

I again repeat the plea that we need to write in ALL preferences into the KLC, so that both contributors and reviewers know what standard is being worked to. It seems we frequently get held up on "philosophy" issues, when time would better be spent improving the many poor quality parts to a basic level of KLC compliance...

Personally, I would prefer a neutral generic part which can be associated with a suitable footprint and MPN, and atomic parts which associated directly with a footprint and MPN, but I would not insist all  possible atomic variations of MPN and footprint are provided. That would be totally impractical considering for some parts there are thousands of variations. We can't hope to include all of them.
If the user does not find the desired part in the atomic parts, he/she can use the generic part as a base, which at least saves them creating a new symbol.

poeschlr, 2018-01-30:
In addition i think what we call atomic is not really atomic as we allow the use of generic footprints for multiple symbols. A true atomic part would have a 1:1 connection between symbol and footprint. (A few relay footprint/symbol pairs would fulfill this requirement.)

Maybe the term "fully specified symbol" would be better?

The thinking now is that we have 4 different possibilities:
 - generic symbol 
     - generic footprint filters. No footprint field.
     - All symbols in the device lib, switch lib and connector_* libs
 - fully specified symbol with generic footprint name suffix
    - default footprint is a generic footprint. Footprint filter set to allow slight variations of this footprint
    - Used where many manufacturers produce a part with differing part number details but compatible 
    - the part uses the base part number followed by a suffix for the package
 - fully specified with mpn as identifier 
    - default footprint is a generic footprint. Footprint filter set to allow slight variations of this footprint
    - I think this is what we currently call "atomic"
    - This uses the MPN but with "unimportant" options replaced by the wildcard character "x"
 - True atomic part
    - default footprint is a specialized footprint for this one part.
    - everything else is equal to fully specified with mpn as identifier

True atomic only makes sense for a very limited set of parts (some relays are an example here)
Using the MPN to fully specify a part only makes sense for parts that are special to one manufacturer. Otherwise the package suffix should be used.
Generic parts are useful for different reasons:
 - It allows a few symbols to "represent" a wide variety of parts
 - They can be used as a style guide for more specialized symbols
Generic parts are only allowed in the libs listed above.

poeschlr, 2018-02-15:
@bobc i have created a PR that should clear up this particular rule: https://github.com/KiCad/kicad-website/pull/272

myfreescalewebpage, 2019-03-18:
@poeschlr and @evanshultz following response of Rene above, I think this issue is answered and can be closed. Your opinion ? Joel

myfreescalewebpage, 2019-03-23:
@poeschlr and @evanshultz ping?

evanshultz, 2019-03-23:
Yes, prefer symbols with a specific footprint and never allow any footprints to be matched that are invalid for the symbol.

myfreescalewebpage, 2019-03-24:
That's also my point of view.
Closing the issue.

closed 2019-03-24 by myfreescalewebpage