KiCad/kicad-symbols #254: Error in Sensor.lib on Place Symbol

aurabindo, 2018-01-30:
I am getting an this error when I try to place a part in eeschema. I am using kicad git 1bc3dfef:

```
Error occurred loading symbol library Sensor.

invalid document library file version formatting in header in input/source
"/home/aj/kicad_sources/kicad_github/kicad-symbols/Sensor.dcm"
line 1, offset 0
```
The sensor library is missing in the selection as well, because of this error.

The specified file contains an weird space on the first line. Git blame points it to c7fe9849.
If I checkout to the next recent commit d60ccf46 on Sensor.lib, I do not get this error on Sensor.dcm (Although another error for not being able to find Timer_PLL library pops up, but I think we can neglect it since this library might have got added between these commits).

Could someone reproduce it ?

poeschlr, 2018-01-30:
My nightly build can open it with no problems, i am still committing a fix for it in https://github.com/KiCad/kicad-symbols/pull/255

closed 2018-01-30 by jkriege2

aurabindo, 2018-01-30:
@jkriege2 @poeschlr  This does not seem to have fixed the issue for me. I created a new project, and launched place symbol tool, and now I have error in line 7 instead of line 1 now.
[image]

An `$ENDCMP` is apparently missing.  Below patch seems to fix it

```
From 8772ed8dcc40ab4576c7abc7d88056022f81503f Mon Sep 17 00:00:00 2001
From: Aurabindo J <mail@aurabindo.in>
Date: Tue, 30 Jan 2018 15:46:16 +0530
Subject: [PATCH] Add missing ENDCMP directive

---
 Sensor.dcm | 1 +
 1 file changed, 1 insertion(+)

diff --git a/Sensor.dcm b/Sensor.dcm
index 1346bf1..563505d 100644
--- a/Sensor.dcm
+++ b/Sensor.dcm
@@ -4,6 +4,7 @@ $CMP ADE7758
 D Poly Phase Multifunction Energy Metering, SO-24
 K Energy Metering
 F http://www.analog.com/media/en/technical-documentation/data-sheets/ADE7758.pdf
+$ENDCMP
 $CMP DHT11
 D Temperature and humidity module
 K digital temperature humidity sensor
-- 
2.11.0
```

poeschlr, 2018-01-31:
Fix merged

commit 0fc25fed (2018-01-30): Remove space from start of file header to fix #254
commit 79acb61e (2018-01-30): Merge pull request #255 from poeschlr/remove_space_from_file_header

Remove space from start of file header to fix #254