KiCad/kicad-symbols #260: Many "small" symbols have wrong pin-label-offset <20mil

jkriege2, 2018-01-31:
- discovered here: https://github.com/KiCad/kicad-symbols/pull/242
- a semi-automated search&replace-run with regexp should solve that problem
- affected libs: transistor_*.lib, device.lib, maybe diode.lib and led.lib, ...

poeschlr, 2018-01-31:
We could also make the script more intelligent. If there is no pin name than the offset for it does not really make sense.

jkriege2, 2018-01-31:
yes, but getting those values to a good default value makes sense to me!

evanshultz, 2018-01-31:
@jkriege2 
I just merged #242. Can you merge and make sure that symbol conforms before I merge this?

Ratfink, 2018-04-18:
I'd prefer to make the symbols follow the existing rules by setting the offset to 20 mil where appropriate.  Yes, the scripts *could* be changed to detect if no pin text is actually rendered and allow 0 offset then, but that would be a whole complicated new code path to maintain, and a new addition to KLC to write and make sure contributors understand, just to avoid the one-time work of adding some 2's to the libraries.

Of course, I consider this a low-priority item that could wait until after 5.0, but I'd just like to add my thoughts.

evanshultz, 2018-04-18:
It would make sense to just have the symbols all be 20mil, but that doesn't work in all cases. Open the below symbol in KiCad, change pin name offset to 20mil, and save. Check the LIB file. It's 1mil again.

I'm not sure why, but I've only seen this on FET symbols. A diode doesn't do it and bigger black-box symbols don't either. I noticed this a while ago but didn't comment on it above.

I also prefer getting everything to 20mil so we meet KLC, but it won't stay at 20mil for all symbols.

It feels like a potential bug in KiCad? Can anybody else confirm? I'll submit it if so.

Application: kicad
Version: (5.0.0-rc2-dev-114-g8fcbb64a4), release build
Libraries:
    wxWidgets 3.0.3
    libcurl/7.54.1 OpenSSL/1.0.2l zlib/1.2.11 libssh2/1.8.0 nghttp2/1.23.1 librtmp/2.3
Platform: Windows 7 (build 7601, Service Pack 1), 64-bit edition, 64 bit, Little endian, wxMSW
Build Info:
    wxWidgets: 3.0.3 (wchar_t,wx containers,compatible with 2.8)
    Boost: 1.60.0
    Curl: 7.54.1
    Compiler: GCC 7.1.0 with C++ ABI 1011

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=OFF
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_SPICE=ON

```
# C3M0030090K
#
DEF C3M0030090K Q 0 0 Y N 1 F N
F0 "Q" 200 75 50 H V L CNN
F1 "C3M0030090K" 200 0 50 H V L CNN
F2 "Package_TO_SOT_THT:TO-247-4_Vertical" 0 0 50 H I C CIN
F3 "" 0 0 50 H I L CNN
ALIAS C3M0065100K C3M0075120K C3M0120100K
$FPLIST
 TO?247*
$ENDFPLIST
DRAW
C 65 0 111 0 1 10 N
C 100 -70 11 0 1 0 F
C 100 70 11 0 1 0 F
P 2 0 1 0 -100 0 10 0 N
P 2 0 1 0 30 -70 100 -70 N
P 2 0 1 10 30 -50 30 -90 N
P 2 0 1 0 30 0 100 0 N
P 2 0 1 10 30 20 30 -20 N
P 2 0 1 0 30 70 100 70 N
P 2 0 1 10 30 90 30 50 N
P 2 0 1 0 100 -70 100 -100 N
P 2 0 1 0 100 -70 100 0 N
P 2 0 1 0 100 100 100 70 N
P 2 0 1 0 145 15 115 15 N
P 3 0 1 0 -100 -100 -100 -150 100 -150 N
P 3 0 1 10 10 75 10 -75 10 -75 N
P 4 0 1 0 40 0 80 15 80 -15 40 0 F
P 4 0 1 0 100 -70 130 -70 130 70 100 70 N
P 4 0 1 0 130 15 115 -10 145 -10 130 15 N
X D 1 100 200 100 D 50 50 1 1 P
X S 2 100 -200 100 U 50 50 1 1 P
X DS 3 -200 -100 100 R 50 50 1 1 P
X G 4 -200 0 100 R 50 50 1 1 I
ENDDRAW
ENDDEF
#
```

Ratfink, 2018-04-19:
I tried on a few other symbols (diodes, ICs) and it worked the same way.  Pin names inside→offset can be changed, ¬Pin names inside→offset always 1.

Looking at the source code, I understand now!  Within the library file, an offset of 0 disables "Place pin names inside," and any other value enables it and uses that offset.  So an offset of 0 should *always* be allowed, because that just means "Place pin names inside" is disabled, and that setting is not forbidden by KLC.  I suggest we change KLC to clarify this, and I'll make a PR to the library-utils to not complain about 0 offset shortly.

Ratfink, 2018-04-19:
Script PR created: https://github.com/KiCad/kicad-library-utils/pull/235

evanshultz, 2018-04-19:
Oh, I think I see what's happening. The UI doesn't allow '0' to be shown. So even if the line of the symbol is `DEF C3M0030090K Q 0 0 Y N 1 F N`, it will display the offset as 1 in the UI if pin names are placed outside.

Ratfink, 2018-04-19:
[KLC]: http://kicad-pcb.org/libraries/klc/ "KiCad Library Convention"
[G1.1]: http://kicad-pcb.org/libraries/klc/G1.1/ "Only standard characters are used for naming libraries and components"
[G1.2]: http://kicad-pcb.org/libraries/klc/G1.2/ "Each library is limited to 250 items"
[G1.3]: http://kicad-pcb.org/libraries/klc/G1.3/ "Libraries are organized by functionality"
[G1.4]: http://kicad-pcb.org/libraries/klc/G1.4/ "English language should be used throughout libraries"
[G1.5]: http://kicad-pcb.org/libraries/klc/G1.5/ "Plural naming is to be avoided"
[G1.6]: http://kicad-pcb.org/libraries/klc/G1.6/ "Capitalization conventions"
[G1.7]: http://kicad-pcb.org/libraries/klc/G1.7/ "Library files use Unix style line endings"
[G1.8]: http://kicad-pcb.org/libraries/klc/G1.8/ "Components can only contain features supported in latest stable release"
[G1.9]: http://kicad-pcb.org/libraries/klc/G1.9/ "Dimensional units"
[G2.1]: http://kicad-pcb.org/libraries/klc/G2.1/ "Defining generic and atomic parts"
[S1.1]: http://kicad-pcb.org/libraries/klc/S1.1/ "Symbol libraries should be categorized by function"
[S2.1]: http://kicad-pcb.org/libraries/klc/S2.1/ "General symbol naming guidelines"
[S2.2]: http://kicad-pcb.org/libraries/klc/S2.2/ "Non-functional variations in part number should be replaced with wildcard"
[S2.3]: http://kicad-pcb.org/libraries/klc/S2.3/ "Where atomic parts are available in multiple footprint options, a separate symbol must be drawn for each footprint"
[S3.1]: http://kicad-pcb.org/libraries/klc/S3.1/ "Origin is centered on the middle of the symbol"
[S3.2]: http://kicad-pcb.org/libraries/klc/S3.2/ "Text fields should use a common text size of 50mils"
[S3.3]: http://kicad-pcb.org/libraries/klc/S3.3/ "Symbol outline and fill requirements"
[S3.4]: http://kicad-pcb.org/libraries/klc/S3.4/ "Symbols with complex functionality may incorporate simple functional diagrams"
[S3.5]: http://kicad-pcb.org/libraries/klc/S3.5/ "Pin connection points must be placed outside of the symbol"
[S3.6]: http://kicad-pcb.org/libraries/klc/S3.6/ "Pin name position offset"
[S3.7]: http://kicad-pcb.org/libraries/klc/S3.7/ "Pin numbering for exposed pads"
[S3.8]: http://kicad-pcb.org/libraries/klc/S3.8/ "Multi unit symbols"
[S4.1]: http://kicad-pcb.org/libraries/klc/S4.1/ "General pin requirements"
[S4.2]: http://kicad-pcb.org/libraries/klc/S4.2/ "Pins should be grouped by function"
[S4.3]: http://kicad-pcb.org/libraries/klc/S4.3/ "Rules for pin stacking"
[S4.4]: http://kicad-pcb.org/libraries/klc/S4.4/ "Pin electrical type"
[S4.5]: http://kicad-pcb.org/libraries/klc/S4.5/ "Pins not connected on the footprint may be omitted from the symbol"
[S4.6]: http://kicad-pcb.org/libraries/klc/S4.6/ "Hidden pins"
[S4.7]: http://kicad-pcb.org/libraries/klc/S4.7/ "Active low pins should be designated using a bar above the symbol name"
[S5.1]: http://kicad-pcb.org/libraries/klc/S5.1/ "Symbols with a default footprint link to a valid footprint file"
[S5.2]: http://kicad-pcb.org/libraries/klc/S5.2/ "Footprint filters should match all appropriate footprints"
[S6.1]: http://kicad-pcb.org/libraries/klc/S6.1/ "Component Reference Designator (RefDes)"
[S6.2]: http://kicad-pcb.org/libraries/klc/S6.2/ "Component fields must be filled appropriately"
[S6.3]: http://kicad-pcb.org/libraries/klc/S6.3/ "Component metadata is added to symbol (and all aliases)"
[S7.1]: http://kicad-pcb.org/libraries/klc/S7.1/ "Power flag symbols"
[S7.2]: http://kicad-pcb.org/libraries/klc/S7.2/ "Graphical symbols"
[F1.1]: http://kicad-pcb.org/libraries/klc/F1.1/ "Footprint libraries are categorized by function"
[F1.2]: http://kicad-pcb.org/libraries/klc/F1.2/ "Connector footprint libraries"
[F2.1]: http://kicad-pcb.org/libraries/klc/F2.1/ "General footprint naming conventions"
[F2.2]: http://kicad-pcb.org/libraries/klc/F2.2/ "Footprint naming field prefixes"
[F2.3]: http://kicad-pcb.org/libraries/klc/F2.3/ "Manufacturer specific version of generic footprints"
[F2.4]: http://kicad-pcb.org/libraries/klc/F2.4/ "Footprint naming for non-standard pin numbering"
[F2.5]: http://kicad-pcb.org/libraries/klc/F2.5/ "Footprint naming conventions for specific components"
[F3.1]: http://kicad-pcb.org/libraries/klc/F3.1/ "SMD chip package naming conventions"
[F3.2]: http://kicad-pcb.org/libraries/klc/F3.2/ "Resistor naming conventions"
[F3.3]: http://kicad-pcb.org/libraries/klc/F3.3/ "Capacitor naming conventions"
[F3.4]: http://kicad-pcb.org/libraries/klc/F3.4/ "SMD IC package naming conventions"
[F3.5]: http://kicad-pcb.org/libraries/klc/F3.5/ "THT IC package naming conventions"
[F3.6]: http://kicad-pcb.org/libraries/klc/F3.6/ "Connector naming conventions"
[F3.7]: http://kicad-pcb.org/libraries/klc/F3.7/ "Fuse naming conventions"
[F4.1]: http://kicad-pcb.org/libraries/klc/F4.1/ "Datasheet recommendations take priority"
[F4.2]: http://kicad-pcb.org/libraries/klc/F4.2/ "Pin 1 should be located at the top left"
[F4.3]: http://kicad-pcb.org/libraries/klc/F4.3/ "Connected copper elements have the same pad number"
[F4.4]: http://kicad-pcb.org/libraries/klc/F4.4/ "Thermal pads"
[F4.5]: http://kicad-pcb.org/libraries/klc/F4.5/ "Specifying footprint keepout areas"
[F4.6]: http://kicad-pcb.org/libraries/klc/F4.6/ "Local clearance and settings should be set to zero"
[F5.1]: http://kicad-pcb.org/libraries/klc/F5.1/ "Silkscreen layer requirements"
[F5.2]: http://kicad-pcb.org/libraries/klc/F5.2/ "Fabrication layer requirements"
[... cut ...]
[M1.3]: http://kicad-pcb.org/libraries/klc/M1.3/ "Source files for 3D models should be supplied"
[M2.1]: http://kicad-pcb.org/libraries/klc/M2.1/ "Permitted 3D model file types"
[M2.2]: http://kicad-pcb.org/libraries/klc/M2.2/ "Model alignment and scaling"

Now that we've cleared up part of the problem, and concluded that [S3.6] needs clarification as to when exactly it applies, I'm fine with adding the additional condition that the rule only applies when ***draw_pinname*** is set to `Y`, which should remove most of the warnings this issue is about.

Ratfink, 2018-04-19:
I've created another PR, https://github.com/KiCad/kicad-library-utils/pull/236, that checks if pin names are drawn before checking the pin name position offset.  This would be a KLC change, so I'd like to see some consensus that this is the right move before merging!

Ratfink, 2018-04-24:
With both https://github.com/KiCad/kicad-library-utils/pull/235 and https://github.com/KiCad/kicad-library-utils/pull/236 merged, I feel like anything the scripts report now could reasonably be considered an actual library bug.  Do we all agree, or is there some other case that the scripts should be smarter about?

One case I see now is e.g. `Device:C`, where the pin name position offset is 10 mil, the pin names are set to be drawn, but neither pin has a name.  I suppose it could be argued that this should be allowed as well, but I consider this to be a library problem (if there are no named pins, why are pin names set to be drawn?).

The only other case I can think of is if all the pins that have names are invisible.  But again, if no named pins are being drawn, why are pin names set to be drawn?