KiCad/kicad-symbols #261: Transistor dcm cleanup

matyro, 2018-01-31:
Currently the description and keywords of Mosfets (and other transistors) has some inconsistencies. A uniform design for the library would be nice.

Here is a list of things that you can find currently in the description [?Added for actual numbers] :

?V Vds
*Mosfet type*
?V Logic Level
?V Vgs
?mOhm Ron
?nC Qg (typ)
*Footprint*

Problems:
- Missing values
- Flipped order
- How should the mosfet type be written? Is N-Channel enough or should manufacture specific extensions like OptiMOS/NexFET/OptiMOS/... be included? Should it be named N-Channel MOSFET, N-Channel FET Transistor, N-Channel power MOSFET? The same question with JFET...
- When should the transition from mUnit to Unit be? (I think from 99mUnit to 0.1Unit)

Next point are the Keywords what should be included there?

antoniovazquezblanco, 2019-06-27:
@matyro could you please make a proposal for both fields?

Thank you!

paulvdhoeven, 2019-07-12:
Drain Current and "Vgs" for P-channel mosfets is also inconsistent.
For about half the P-Channel (MOS)Fets these are positive in the descriptions, while for the other half these are negative (which is more common).

For an overview:
Eeschema / 'a' , and type: "p-channel" in the search box.
See also thread and 2nd screenshot at:
https://forum.kicad.info/t/p-channel-mosfet-symbols-and-footprints/18012/8?u=paulvdh