KiCad/kicad-symbols #272: Various library errors

evanshultz, 2018-02-02:
From https://github.com/KiCad/kicad-symbols/pull/269:
- Sensor_Magnetic.lib:AS5048*: bad FPfilter
- Reference_Voltage.lib:REF5020ADGK, Reference_Voltage.lib:LT6657AHMS8-2.5: FPfilter should be `MSOP*3x3mm*P0.65mm*`
- Memory_RAM.lib:AS6C1616: FPfilter should be `TSOP?I*18.4x12mm*P0.5mm*` or just `TSOP*18.4x12mm*P0.5mm*`
- Potentiometer_Digital.lib:DS1267_SOIC: bad text sizes and FPfilter missing trailing `*` and contains extra fields
- MCU_Microchip_PIC18.lib:PIC18F13K50-ESS: FPfilter should be `SSOP*5.3x7.2mm*P0.65mm*`
- The above also caught that the footprint `Package_SO:SSOP-20_5.3x7.2mm_P0.65mm` is named wrong and should be `Package_SO:SSOP-20_5x3x7.2mm_P0.65mm` (period is an `x`)
- Memory_EEPROM.lib:AT24CS*: FPfilter should be `SOIC*3.9x4.9mm*P1.27mm*` and pins are off grid (maybe Bob will be fixing this?)
- Memory_Flash.lib:AT45DB161-RC: FPfilter sis missing and pins are off grid (maybe Bob will be fixing this?)
- Timer_RTC.lib:DS3231: Wrong stacked pin names
- Regulator_Linear.lib:AP2204MP-ADJ: footprint and FPfilter mismatch
- Interface_USB.lib:FT240XS: VCC and GND pins are bidirectional type
- Interface_USB.lib:FT221XS: FPfilter is apparently an ALIAS
- Power_Protection.lib:USB6B1: has two FPfilters with a default footprint
- MCU_ST_STM32.lib:STM32*: many are missing FPfilter
- Analog_DAC.lib:MC1408_SOIC: FPfilter should be `SOIC*3.9x4.9mm*P1.27mm*` (9 instead of 4)
- MCU_Atmel_8051.lib:AT89*2051-SU: Loads of off-grid pins and no FPfilter
- Power_Management.lib:LM5051, Power_Management.lib:LT1641-1: FPfilter should be `SOIC*3.9x4.9mm*P1.27mm`
- Regulator_Switching.lib:MIC2177, Regulator_Switching.lib:MIC2178: missing FPfilter
- Regulator_Switching.lib:ADP5070AREZ: FPfilter should be `ETSSOP*4.4x6.5mm*P0.65mm*`
- MCU_Atmel_AVR.lib:AT90PWM1-16SU: missing FPfilter
- MCU_Atmel_AVR.lib:AT90Sxxxx-4SC: many off-grid pins and missing FPfilter
- Analog_ADC.lib:ADS128D818: off-grid pins
- MCU_Atmel_ATTINY.lib: many symbols with off-grid pins and/or no FPfilter
- MCU_Microchip_PIC12.lib:PIC12LF1840T48-IST: appears to come in only TSSOP14 but FPfilter should be the default footprint and the default footprint is just wrong
- Driver_LED.lib:HV9961NG-G, Driver_LED.lib:HV9931LG-G, Driver_LED.lib:HV9961LG-G, Driver_LED.lib:HV9930LG-G: VIN pin should be Power Input type and multiple FPfilters with default footprint
- Driver_LED.lib:HV9972LG-G, Driver_LED.lib:HV9925SG-G: multiple FPfilters with default footprint
- Analog.lib:CAP002DG: needs FPfilter updated due to the footprint apparently being moved to a different library
- Transistor_FET.lib:IRF7404, Transistor_FET.lib:FDS6890A, Transistor_FET.lib:IRF8721PBF-1, Transistor_FET.lib:IRF7607PBF, Transistor_FET.lib:IRF7343PBF, Transistor_FET.lib:IRF7309IPBF, Transistor_FET.lib:FDS9934C: off-grid pin(s)

strombom, 2018-02-02:
See pull request: https://github.com/KiCad/kicad-symbols/pull/275

I have tried to fix most of the issues above with those exceptions:

* Timer_RTC.lib:DS3231: Not sure how to solve this
* MCU_ST_STM32: Since those are script generated maybe the script should include footprint filter
* MCU_Atmel_ATTINY: A bit too many to fix now

bobc, 2018-02-03:
I've addressed Memory_EEPROM.lib issues in #222

Misca1234, 2018-02-16:
Timer_RTC.lib is addressed in 
https://github.com/KiCad/kicad-symbols/pull/307

evanshultz, 2018-02-16:
@bobc 
Thanks!

@Misca1234 
The checks above correspond to me PR at https://github.com/KiCad/kicad-symbols/pull/276, and also includes that fix.

evanshultz, 2018-02-28:
https://github.com/KiCad/kicad-symbols/pull/276 and https://github.com/KiCad/kicad-symbols/pull/332 handle all but the memory and MCU issues.

daab4848, 2018-03-09:
@evanshultz Do you know any PRs that are fixing the leftover issues from the above list? I wanted to pick a bone with one or a few of these issues.

Ratfink, 2018-03-09:
#305 adds the missing FPFilters to the STM32 symbols.

evanshultz, 2018-03-09:
@daab4848 
Thanks for volunteering! I believe fixing the Memory_* libraries should be valuable. I don't know if anybody is fixing them in the short-term, so if you want to check all those libraries locally and make fixes that would be great. Longer term I believe the memory libraries will be script-based. But that shouldn't stop us from fixing other errors now.

The Atmel libraries are, I believe, almost fully fixed with regards to footprints and FPfilters. Having that done makes it easier to do further changes because there's less chance of having branch conflicts. I believe https://github.com/KiCad/kicad-symbols/pull/340 is the last of those PRs, and so once that PR is merged you shouldn't have issues making fixes there.

Also, if you want to run the check scripts on the entire library you're welcome to fix anything you find. Surely there are still lots of symbols (and footprints, too!) which need work.

evanshultz, 2018-03-09:
@Ratfink 
Great! I checked that item above since I think it's safe to assume once a PR is submitted it will be merged. That also helps to leave only items without an "owner" as unchecked above.

daab4848, 2018-03-10:
See PR KiCad/kicad-symbols #366 for the Memory_Flash.lib:AT45DB161-RC fixes (short-term)

daab4848, 2018-03-15:
@evanshultz See https://github.com/KiCad/kicad-symbols/pull/382 for MCU_Atmel_8051.lib:AT89xxxx-xx fixes

Ratfink, 2018-04-24:
All items here are checked.  @evanshultz should this be closed now?

evanshultz, 2018-04-24:
Yep. It seems you made the last few edits to the list, so I suppose that means the open items I last saw are closed. Thanks for the reminder!

closed 2018-04-24 by evanshultz