KiCad/kicad-symbols #298: US-style resistor

jkriege2, 2018-02-12:
It was discussed over in the forums several times that many US-people wish for a "wiggly" resistor symbol (IEC prefers the box-symbol).

There was the idea to add these as alternative via the deMorgan-alternative-function in the symbol editor (I don't think we really use it for deMorgan-version of the logic gates).

So the question is: 
1. Should we do that at all?
1. And if so:
  a. only the basic resistor + pots
  b. only the basic resistor + pots
  c. basic resistors, pots, sensors (NTC, PTC, LDR ...)
  d. all, including resistor networks
1. If we do it: How should we handle new contributions (also internal schematics of ICs and transistors ...): 
  a. box-required, wiggly as an additional option
  b. either box or wiggly 
  c. both

I would say we start with 2a and require 3a (just my personal preference)

best,
JAN

bobc, 2018-02-13:
2a and 2b seem to be identical?

The logic gates I generated have de Morgan equivalents, I think other libs such as 74xgxx do not.

We should probably think about how to handle alternative shapes in the proposed SWEET format, and what changes to make in KiCad. For example, as a user, I might want to select a "US style" theme, and thus get the US style as default (instead of having to click each part and select "Converted shape").

Currently we have different libs for ANSI style logic gates and IEC style gates, that might a better way to manage it in the short term.

poeschlr, 2018-02-13:
I am not sure it is worth our effort. I think there are more important issues to take care of. 
These changes only include artistic improvements for users who prefer the us style. (No user in the thread that prompted this issue complained that they are unable to read a schematic with the box style resistors. They complained that they like the other type more.)

And i fear if we add a us style for some symbols users will start to complain why only for these and not for all others. The next complained will be why the European style is used as the default, ....
(And yes i was more optimistic about the community back when this original discussion took place. That's what being a lib maintainer for a year or two did to me.)

---

Just to clarify: I am not opposed to it, but i think other issues would be more important. (I would give nearly other open issue in any of our repos a higher priority.)

pointhi, 2018-02-13:
My position on this issue would be: 
1.) Waiting for the new symbol format, and trying to get a feature supporting this usecase into it
2.) Changing the KLC to something like: IEC is mandatory for every symbol, and other styles are optional

poeschlr, 2018-02-13:
@pointhi the us style resistor is also in the IEC standard. (At least in the standards that i have access to.)

Ratfink, 2018-04-12:
My feelings on this matter, now that we have #480:

1. Sure, why not?  We already have alternate diode styles as separate symbols, and the world hasn't ended; zigzag resistors shouldn't be any more trouble, I'd imagine.
2. 2.d. is done in #480, so we might as well go with that.
3. 3.a. sounds like the right approach to me (most similar to what we've done with diodes).

I don't see any reason to wait for the .sweet file format, but if new features will make this nicer once we have .sweet, we can address that then.

evanshultz, 2018-04-12:
Playing devil's advocate...
- The diode symbols are only in this lib, but not elsewhere that I see. I've seen librarians request the non-ALT style for diodes in other libs and so have I. So what about resistors that are inside other symbols? Do we use box resistors everywhere but device.lib? Does it make sense to have digital pots with both resistors styles, for example?
- We have a zener symbol with one short line normal to the big line, but the US style I see most commonly is to have [two angled lines](https://www.khanacademy.org/science/electrical-engineering/ee-semiconductor-devices/ee-diode/a/ee-diode-circuit-element) on the big line. (This goes along with the different polarized cap symbols Jan mentioned above, and fuses is another example.)
- Is a suffix of `_US` the best way to name them? It seems egotistical, especially if the wiggly style is in an international spec like IEC or IEEE. Plus "us" is a word in English. And the diodes are suffixed `_ALT`. Naming wiggly resistors `R_ALT` has some intrinsic preference but it's more neutral than `R_US`.

Personally I think it's best to have an accepted wiggly resistor in our library that other users can build from so I'd support it from that standpoint if nothing else.

Also, from the IEC doc I have photoresistors and attenuators only show a wiggly symbol. It's kinda mixed up because not even that spec shows box and wiggly for all resistive symbols.

@jkriege2 
Are 2a and 2b supposed to be identical? I'm with Bob by being confused about that.

jkriege2, 2018-04-13:
Hi!

I think 2a should have been only basic resistors (+pots in 2b) ... I changed it accordingly above!

cu
JAN

Ratfink, 2018-04-19:
#480 has been merged, any reason to keep this open?

jkriege2, 2018-04-20:
closed via fix #480

closed 2018-04-20 by jkriege2