KiCad/kicad-symbols #300: Atmel (ATMEGA) MCU symbols have the wrong footprint assigned and wrong footprint filters

poeschlr, 2018-02-14:
Especially most ATMEGA symbols have footprints with very old footprint naming assigned (no body size, pitch, ... in the name)
They sadly also do not use wildcard chars in the footprint filters but simply copied the footprint field. (So not even the footprint filter will help the user out here.)

Originally reported over at the forum in: https://forum.kicad.info/t/no-footprint-for-tqfp-32/9625

bobc, 2018-02-14:
ATMEGA also has problems with too long pin names. It looks like the footprint name is copied from source data. I guess they were generated by script, does anyone know?

poeschlr, 2018-02-14:
The long pin names are a different topic (and covered in a different issue.)

The reason i separate these two is that the pin name stuff is a KLC issue whereas this is a bug that prevents the symbols to be used as intended.

bobc, 2018-02-14:
Ok, that makes sense. If there was a script to generate the symbols, it seems to be lost in the mists of time, the symbols appear in github in 2008 and I can't find anything earlier.

I think it will have to be done from datasheets.

fauxpark, 2018-03-01:
I have started some work on this in #335 :)

fauxpark, 2018-03-11:
The merge of #340 has resolved this issue (I should have linked it so that it autoclosed).

closed 2018-03-12 by poeschlr