KiCad/kicad-symbols #301: Ethernet connectors

evanshultz, 2018-02-14:
@poeschlr @jkriege2 
I see RJ45 symbols at https://github.com/KiCad/kicad-symbols/blob/master/Connector_Specialized.lib. Is this the correct/desired place for them?

There are a few footprints at https://github.com/KiCad/kicad-footprints/tree/master/Connector_RJ.pretty. Again, is this is the desired place for them, grouped by connector type ("Connector_RJ") rather than function ("Connector_Network" or "Connector_Ethernet")?

I assume a Connector_Ethernet or Connector_Network or Connector_RJ library would be worthwhile if we had enough symbols to warrant a new library. Frankly, it seems to me like we do. This type of connector is common and would seem to be on par with USB, HDMI, etc. Would it be best to have a symbol and footprint repo for RJ connectors?

jkriege2, 2018-02-14:
About the footprint-location: These types of connectors are also used for other purposes, e.g. the old phone-jacks (used on the modem/telephone-side) was often one of these (narrower) RJ/Western-connectors. Also I think some people use the narrow (4-6-pin-version) for I2C ...

Therefore I would keep them together in one Connector_RJ.pretty ... the other solutions would leave us with the question: Where to pu the versions that are not used for Ethernet/Network?

jkriege2, 2018-02-14:
The same we do for e.g. SUB-D ... they are also grouped by type, not function

poeschlr, 2018-02-14:
I don't think we should split the connector symbol libs too fine grained.
As we use generic symbols to represent connectors right now. This would result in one or two symbols per lib if we would split it the same way as the footprint libs.

The only thing i am a bit unsure of is where to put the male/female single row connectors. They are currently in connector specialized, maybe they would fit better into connector general

---

If we however want fully specified symbols for such network connectors then it is a different story.

---

About footprint location: Place footprints under the lib for the original function intended for these connectors. So even if some people use a connector for I2C but it was originally devised for ethernet then put it together with other ethernet connectors.

---

For the future please separate discussion about footprint and symbol lib. (It took me way to long to realize the main discussion here is about footprint libs and not symbol libs)

evanshultz, 2018-02-14:
I ask because of Wuerth 74995151120:http://katalog.we-online.de/pbs/datasheet/7499151120.pdf.

Symbol:
[image]

Footprint:
[image]

We have no specific ethernet connector symbols and only one specific Amphenol footprint that I can find. The rest, both schematic and PCB, are generic and lack datasheet or other supporting info.

But Connector_Specialized.lib is a catch-all term that seems non-obvious to me and houses a lot of different symbols. So where should this new symbol go? Even if there is no better place now, this symbol lib seems like it could be improved with regards to organization. And the name, at least to me, doesn't communicate well to me what symbols I will find inside. I agree not to go too fine-grained, but there are a number of USB symbols and also the single-row connectors you mentioned. Also note that the Interface and Sensor symbol libs (among others) are split up by function.

We already have Connector_RJ.pretty so the footprint is straightforward.

@poeschlr 
Sometimes your tone is really grating to me. My question _does_ revolve around the symbols; I only brought up footprints to contrast their categorization with the relevant symbols.

poeschlr, 2018-02-14:
Your question is one sentence about symbols and two full paragraphs about footprints. (That was the reason why my answer was more about footprints then about symbols. That was also the reason why i requested to separate such things in the future.)

Just remember that i might not understand your questions the same way as you intended them to be understood.

----

About splitting the lib:
I agree that the current names are not optimal. 
I fear splitting the libs will result in 20 or so new libs with < 10 symbols in each lib. That will make neither our life nor the life of users easier.

Additionally as most connectors are not really limited to a specific function, splitting connector symbols by function becomes quite hard to do. (As already hinted above)

Now what can i come up with (For now):
One option would be to drop the Specialized suffix and have a lib called Connector. (Not sure this is better)

Another option would be a connector_standardized_shape holding stuff like sub-d, terminal blocks, (maybe even male/female as these could be seen as limiting the shape)
Plus one for connector_standardized_function which would then hold hdmi, usb, ...

---

Also remember the v5 release is not that far away. (rc1 might be tagged this weekend) Any larger reorg should happen either before that release or wait till the v6 release. (This is the reasons why i am skeptical about large scale changes suggested lately.) 

It has been requested by the packagers (and also some developers and users) to keep the lib as stable as possible during the release cycle. [This was one of the relevant discussions about that on the mailing list](https://www.mail-archive.com/kicad-developers@lists.launchpad.net/msg27862.html) (I faintly remember one where the question was about how the tags should look like that basically was a similar discussion. But i can't find it right now.)

The alternative is that all v5 releases will get the same lib as the first v5 releases. (Not sure if we want that.) But even if the lead developers decide to go that route anyway we might want to make it easy for users to switch over to newer versions that have more content. (Or at least make it easy for users to share their project with users who have an older version of the lib installed.)

evanshultz, 2018-02-14:
I vote for a symbol lib called just `Connector`, moving all symbols from `Connector_Specialized`. This matches having a "base" library which is a dumping ground for random symbols, with underscore-separated libraries for subdivisions. For examples, see Analog, Interface, CPU, etc.

I think that's relatively easy to do and I will do it when I submit the above symbol if y'all agree.

poeschlr, 2018-02-14:
Ok for me.

jkriege2, 2018-02-15:
fine with me

Ratfink, 2018-04-20:
With #310 finally merged, I'm 99% sure this issue is done.  Someone re-open if I'm wrong. 🙂

closed 2018-04-20 by Ratfink