KiCad/kicad-symbols #317: X capacitor discharge IC library location

evanshultz, 2018-02-17:
@poeschlr @jkriege2 
Where should these ICs go? Their purpose is to quickly discharge capacitance stored in X capacitors when AC input voltage to a product is removed.

CAPZero parts are in the Analog library. These are Power Integrations parts which have been in our library a long time. See https://www.power.com/sites/default/files/product-docs/capzero_family_datasheet.pdf. They aren't analog parts IMO.

HF81 and TEA1708T were added later and merged into Power_Management (https://github.com/KiCad/kicad-library/pull/1790) and I think that's the correct library. Should the CAPZero parts be moved to Power_Management?

jkriege2, 2018-02-17:
I fear I'm not really helpful on this:
I'm happy with both places, as I see arguments for both (don't really have a preference ...)

evanshultz, 2018-02-17:
Analog doesn't apply, to me, because these are back-to-back FETs with a controller. There is at least a counter inside them. and a counter is a fundamental digital circuit.

poeschlr, 2018-03-07:
Power management seems like a reasonable place to put them.

If this can be done before rc2 then they should be moved. After rc2 i think we should limit ourselfs to fixing klc issues, fixing bugs and adding new things. 
Major reorganization tasks should then be put of until we branch in preparation for v6. (I think we should give users as stable a lib as possible. Even if this limits some of the things we might want to do.)

evanshultz, 2018-03-07:
Sure! PR submitted at https://github.com/KiCad/kicad-symbols/pull/356. And I just noticed that I forgot to use my own fork. Oops!

closed 2018-03-21 by evanshultz