KiCad/kicad-symbols #329: migrating to v5 libs

diggit, 2018-02-25:
Hi, I want to migrate to latest libs and I noticed, that many symbols were migrated to different libs. There are some .json files which describes probably all those migrations.

Is there any simple way to use those .json files to update symbol names in my .sch files?

poeschlr, 2018-02-25:
The json file contains most migrations but not all of them. I do not know of a script that does this for you. You could re-purpose parts of the symbol moving script found in the kicad-library-utils repo.

Another thing to keep in mind: We did not only move/rename the symbols. Some of them have other changes. (Example: Logic symbols now have a separate power unit and their size did change.)

diggit, 2018-03-12:
Ok, I've done migration by hand. `sed` did most of the job.

closed 2018-03-12 by diggit