KiCad/kicad-symbols #352: Nordic Semi MCU organization

evanshultz, 2018-03-05:
I find symbols at https://github.com/KiCad/kicad-symbols/blob/master/RF_Bluetooth.lib. Another symbol that hasn't been merged is at https://github.com/KiCad/kicad-symbols/pull/75.

However, they should be in an MCU library. I do not see one for this manufacturer present, but we should create MCU_Nordic.lib to hold them once the un-merged PR above is resolved.

Ratfink, 2018-03-06:
If we're moving general-purpose MCUs with built-in radios to the MCU libraries, there's more to move than just Nordic Semi parts.  Off the top of my head:

* RF_WiFi.lib: ESP8266EX → new library MCU_Espressif.lib

And possibly:

* RF_Wifi.lib: ESP-12E → MCU_Module.lib
* RF_Wifi.lib: Particle_P1 → MCU_Module.lib

evanshultz, 2018-03-07:
Right. Even if there are BT and/or WiFi and/or other peripherals, the core function of the part is an MCU so it should be in a library that is obviously for MCUs.

The above all sounds fine to me, and thanks for checking it out. Normally I'd say we should have a reasonable number of symbols to support a new library. But seeing as how we don't have a vanilla MCU.lib, perhaps all MCU are to be segregated by manufacturer regardless of the number of symbols?

evanshultz, 2018-03-12:
A few more inspired by https://github.com/KiCad/kicad-symbols/pull/349.

Move to RF_Module:
- MM002 (and update URL)
- TD1205
- TD1208
- UL865 (and update URL)
- iM880A
- iM880B

Move to RF_Amplifier (amplification and attenuation are essentially the same):
- MAADSS0008

evanshultz, 2018-03-13:
PR submitted at https://github.com/KiCad/kicad-symbols/pull/379.

closed 2018-04-19 by Ratfink