KiCad/kicad-symbols #353: Should we add a Code of Conduct?

Ratfink, 2018-03-06:
There have recently been some inflammatory, condescending, and generally negative remarks on at least one pull request.  It's always a shame to see bad behavior seep into a project, and I'd really hate to see this sort of negativity persist and ultimately harm the quality and/or quantity of work done on these libraries.

In light of this, I would like to propose that we establish a formal code of conduct for the KiCad libraries.  Perhaps this is actually a discussion we should have with the development team as well, so that the KiCad project as a whole may benefit from this addition.

stambaughw, 2018-03-06:
This is one of the reasons the kicad developers mailing list remains on 
launchpad and is a closed mailing list.  No one should tolerate abuse! 
Everyone is entitle to their opinion.  They are not entitled to abuse 
other people.  I don't know if you can boot people from the mailing list 
on github but if you can, it's the only effective tool in your toolbox 
to stop this behavior.  My problem with codes of conduct is that if you 
actually have to tell people not act like children, then a code of 
conduct probably isn't going to make much of a difference.

poeschlr, 2018-03-07:
I assume this is inspired by the discussion in https://github.com/KiCad/kicad-symbols/pull/280
Yes it contains a heated discussion. (Which is ok. For some things there is no clear right or wrong way.) 

Sadly some comments in there can be interpreted as personal attacks. I assume they are not meant that way. (The latest comment by @evanshultz seems to back me up on that.)

---

If we want to add a code of conduct i think it should simply state that we expect everyone to tread each other with respect. And that personal attacks are not welcome. Maybe we can hint at the fact that discussions are allowed (and maybe even encouraged) as long as they stay civil and on topic.

I think the only tool available to us is to delete comments if they are abusive. I am not really comfortable to use this tool lightly. I think it should be a last resort for repeated offenses. (I really hope it will never get this far.)

Ratfink, 2018-03-07:
It is possible for a GitHub organization's owners to block users from collaborating on any of the organization's repositories.  See the help article [here](https://help.github.com/articles/blocking-a-user-from-your-organization/) for details of how to do it and what exactly it blocks.  It's also possible to [lock individual issues/pull requests](https://help.github.com/articles/locking-conversations/) so that only members of the organization can write in them, which is a much less harsh tool than blocking users or deleting comments.

Wayne, I understand your point about a code of conduct not making people behave well on its own.  No matter what, most people will behave well and a few will behave badly.  I don't see that as the purpose though.  As I see it, the code of conduct would mainly serve to remove ambiguity, making it clearer what the limit of acceptable behavior is and when maintainers should take action (and just as importantly, when we should not).