KiCad/kicad-symbols #360: TRANSISTOR_FET lib: CSD17578Q5A wrong footprint

greenscreenflicker, 2018-03-08:
SON8 as footprint, however footprint is SON 5 mm × 6 mm 
http://www.ti.com/lit/ds/symlink/csd17578q5a.pdf

Ratfink, 2018-03-08:
I just had a look, and it seems the footprint assigned to this transistor in the library is very close to, but not exactly the same as, the footprint recommended by TI.  It'll probably work as-is, but if someone would like to add the exact footprint TI recommends, that would be welcome!

evanshultz, 2018-03-08:
@greenscreenflicker 
Would you be willing to create the correct footprint? If so, we can help you get it added to the footprint library and make it the default footprint for this transistor.

firebladed, 2018-06-13:
I think that at some point they changed recommendations to use a q5B which is backwards compatible footprint https://e2e.ti.com/support/power_management/power_stage/f/208/t/602226?jktype=e2e
which i think is a TO_SOT_Packages_SMD:TDSON-8-1 is that a better choice ?