KiCad/kicad-symbols #381: What pin type for analog pins?

evanshultz, 2018-03-14:
@poeschlr @jkriege2 @Ratfink @Misca1234 
A few discussions recently have been about pin types. And I'm not clear what the KiCad library should do with analog pins (pins that do not generate or recieve a logic level or digital signal) which don't have an obvious function of input or output.

https://github.com/KiCad/kicad-symbols/pull/322 is one example regarding the MULTOUT pin of UC3854 (http://www.ti.com/lit/ds/slus336a/slus336a.pdf). This pin is an output of some internal circuitry and also the input to some internal circuitry. It does not send or receive signals to any other ICs.

It also only nominally connects to resistors or maybe a capacitor. (Of course some user may have a creative design and not adhere to the typical application of this IC.)

Is Bidirectional the right type for this pin? Information can be read on this pin using an oscilloscope, and the circuitry surrounding the IC does input a signal on the same pin. So there is some sense of bidirectionality. However, if Bidirectional pins in KiCad are meant to be used for logic level pins for the purposes of ERC, then this pin may best be set to Passive type.

What should the KiCad library require in this case? What pin type should be used for this, and similar, pins?

Ratfink, 2018-03-14:
I can see valid reasoning for calling it either Bidirectional or Passive.  I'm not sure which would be better, so I'll leave it to our more experienced librarians.

evanshultz, 2018-03-14:
The differences in KiCad should be captured by the ERC options:
[image]

A pin of type Bidirectional generates a warning when connected to a pin of type Power Output or Open Emitter. A Passive pin will not generate any warnings in those cases. Otherwise they are the same to the ERC engine.

Otherwise, I think it's a philosophical question of how the librarians want pin types to be defined/used.

Shackmeister, 2018-03-14:
I think the passive type makes the most sense.

Its a slightly unlucky title when checking the emails on my phone :P
[image]