KiCad/kicad-symbols #387: Fixing Atmel (now Microchip) MCU libs

evanshultz, 2018-03-16:
@fauxpark has done a ton of work already. Here is what is left to consider.

 - [ ] Shrink and align the boxes
 - [ ] Move obsolete symbols to obsolete/
- [ ] Pin types for all pins, especially the special-function ones that aren't power or GPIO.
- [ ] Pin placement when you shrink symbols. Should all GPIO be on the left side, right side, distributed as evenly as possible, etc.
- [ ] Do all XTAL pins go on the same side? How far apart are they so a crystal symbol fits there nicely.
- [ ] Pin text offset should be 20mil.
- [ ] Place power and ground pins next to each other or leave a space between them (like the GND and EP pins, for example).
- [ ] Pin length should be 100mil if pin numbers have <3 characters (usually 99 pins or less). All NC pins can be 100mil.
- [ ] Place refdes and MPN inside the symbol outline if it's large enough?
- [ ] Renam AT89S2051/S4051-PU and AT89S2051/S4051-SU symbols to have a "24" in them (datasheet indicates that this is the clock speed i.e. the AT89Sxxxx versions run at 24MHz). So rename as AT89S2051/S4051-24PU and AT89S2051/S4051-24SU.
- [ ] Review/fix description, keywords, and datasheet for all symbols.

See https://github.com/KiCad/kicad-symbols/pull/373 and https://github.com/KiCad/kicad-symbols/pull/382 for some previous discussion on those points. To keep it all together, though, let's move all discussion here.

fauxpark, 2018-03-17:
The pin name offsets should already be done.

RE renaming the 8051 symbols  - should we just remove that bit from all symbols instead, or add more aliases for clock speed? Right now it seems there is mostly just one set of symbols defined for each device.

evanshultz, 2018-04-16:
@fauxpark 
Now that https://github.com/KiCad/kicad-symbols/pull/389 has been merged, I checked off a few more items from the list above. Please confirm that the checked off items are correct, and also if there are any more items to add. If you're willing to keep going it would be great!

fauxpark, 2018-07-18:
Second point (obsolete symbols) is finally done 👍  thanks @poeschlr.

I'll look at pin types next.

evanshultz, 2018-07-19:
Just merged https://github.com/KiCad/kicad-symbols/pull/702, which counts towards the 4th item above.

Elijahg, 2018-09-15:
Thanks for keeping the library up to date, but what was the reasoning behind removing the pin descriptions from the schematic symbol? It was extremely useful to have the various pin functions on the symbol to save hunting through the datasheet and cross-checking for each pin. This seems a step forward in neatness, but a huge step back in terms of functionality & ease of use. I'm going to revert to the previous library for now because of this.

Elijahg, 2018-09-16:
I completely agree symbols do get pretty massive for larger chips. That said, the length of the symbol won't change even with pin descriptions included, it only makes it a bit wider. Perhaps it would be better to keep it on the smaller chips and have a cutoff at say ~50 pins?

[image]

Perhaps a better option would be mousing over particular pins showing a tooltip with the peripherals described, obviously this would require support in Eeschema. I think Fritzing(!) used to do something similar, though only for the port identifier.

poeschlr, 2018-09-16:
The future plan for kicad is to have pin alternatives where one can have alternative pin names and electrical types for each pin.
Until this is implemented (Might not be before version 7) i agree with the way @fauxpark described. (He was not the first to go that way. @hackscribble started this with his rework of the NXP libs. This is at least a year ago if not two.)

Razyre, 2018-09-17:
I think the concept of the tooltips would be quite a sensible way to see extended pin descriptions to save on sheet space.

evanshultz, 2018-10-04:
@fauxpark 
Is item 4 now checked off?

It looks like the 5th and 6th items will break symbols so they can't be done now. Is there anything left? Are there pin names that should be simplified now, or save the effort and wait until symbols can be shrunk (my preference)?

evanshultz, 2018-10-04:
Yes, if the pin position changes at all we'll consider it a breaking change since the schematic will lose connectivity if the symbol is updated.

I would also support a small subset of alternate pin functions, but we'd want to come to a consensus about what to add. And if it makes sense to invest the time at the point when we would get started before the glorious new file format is planned to be released. (Of course, we should add about 1 year to the planned KiCad release date to be in the ballpark of reality.)

I added another item above, to remove the parts of the MPN that we are currently considering extraneous.

jaxter184, 2018-10-20:
Looking through the library, it seems some of the ATtiny chips are missing. The one in particular that I was looking for was the ATtiny404, but I also don't see the ATtiny412 or ATtiny417. Were these excluded on purpose? Or are they under a different name perhaps?

Also, not sure if there's a designated place for this type of question, but while I'm already here: how do I go about contributing to the KiCAD libraries? Is there some sort of style guide for symbols and footprints?

fauxpark, 2018-10-20:
> Looking through the library, it seems some of the ATtiny chips are missing. The one in particular that I was looking for was the ATtiny404, but I also don't see the ATtiny412 or ATtiny417. Were these excluded on purpose? Or are they under a different name perhaps?

I'm getting to them 😉  They weren't excluded, no one has submitted them yet. I've been working to improve the AVR libraries and I'm just getting to the adding-new-symbols stage now.

> Also, not sure if there's a designated place for this type of question, but while I'm already here: how do I go about contributing to the KiCAD libraries? Is there some sort of style guide for symbols and footprints?

Yes, these are called the Kicad Library Conventions (KLC). You can have a look at it here: http://kicad-pcb.org/libraries/klc/.

evanshultz, 2018-10-29:
Added more items to the list in the first comment from https://github.com/KiCad/kicad-symbols/pull/1093.

fauxpark, 2018-10-30:
Hmm...

> Remove MPN suffixes the current library considers extraneous, such as packaging and lead finish.

Lead finish can definitely go, but for these symbols, different packaging sometimes means different pinout, and even if TQFP and QFN have the same pinout (as in the mega32U4 and many others), there is still the extra pin on the QFN for the EP. Thus the packaging code is pretty much never extraneous here.

poeschlr, 2018-10-30:
I think there is a misunderstanding. The package in which the part comes should stay encoded in the MPN (or added as cleartext to the base part number if there is no clear encoding for the package)

However there is often a second packaging option. One level up. Some manufacturers have different order numbers depending on how many parts you with one unit. (So do you get a tray of 100, a reel of 1000 or a single piece. Sometimes the number of parts per unit is the same but the container is different.)

fauxpark, 2018-10-30:
I forgot about that. The only alternate packaging option is -R for tape & reel, which none of the currently submitted symbols include.

antoniovazquezblanco, 2019-02-08:
Checked item number 18 in the main list due to the merge of https://github.com/KiCad/kicad-symbols/pull/1507

chschlue, 2019-09-13:
I took the liberty and added task 21

We currently have _ATmega, _ATtiny and _AVR which consists of AT90s. All of these are AVRs.