KiCad/kicad-symbols #390: Alias ... not found in library

greyltc, 2018-03-17:
I'm using the stable release tagged 5.0.0-rc1
and I get
```
02:39:44 PM: Alias 'XC3S200AN/FT256' not found in library:

'/usr/share/kicad/library/FPGA_Xilinx.dcm'

at line 34 offset 21
02:39:45 PM: Alias 'LM334Z/LFT1' not found in library:

'/usr/share/kicad/library/Reference_Current.dcm'

at line 39 offset 17
```
when the libraries are loaded.

Can we maybe get an rc2 tag with that stuff fixed?

greyltc, 2018-03-17:
b72d6d68fa28868ba52e1b1a5b5883474ed370a8
is a fix for the first one

Ratfink, 2018-03-17:
We can certainly fix the bug!  No symbols are supposed to have slashes in their names, and that's almost certainly the source of the error messages.  One of those problems has already been fixed in master as you pointed out, the other I'll submit a PR for shortly.

We're not in charge of when the KiCad developers decide to tag 5.0.0-RC2, though.  AFAIK our tags just follow theirs.

greyltc, 2018-03-17:
RC2 is out, right?

greyltc, 2018-03-17:
Nevermind! that was an `rc2-dev` tag I was confused by

poeschlr, 2018-03-17:
You are not the only one confused by that tag

greyltc, 2018-03-17:
```
sed -i 's,LM334Z/LFT1,LM334Z-LFT1,' Reference_Current.dcm
sed -i 's,LM334Z/LFT1,LM334Z-LFT1,' Reference_Current.lib
```
seems to fix the second error here

tijuca, 2018-03-17:
Making a new RC isn't hard linked to release candidates of the main applications as long you don't create a "reserved" RC number like `rc2`. Given the problems the current development people need to resolve before RC2 of KiCad5 will be possible it would be good to consider of creating one or more new RC releases of kicad-{symbols,footprints,packages3d} etc. I'd do a `rc1.1` or `rc1.2` and so on if wanted. I also would follow that and create new uploads to Debian.
I guess Wayne will not have hard concerns about this, if unsure it's just a question on the ML.

greyltc, 2018-03-17:
I think that would be great. That way there would be a stable tag without these bugs for me to package (in Arch Linux).

PR #391 (2018-03-18): Remove slash from name of LM334Z-LFT1
Fixes #390.

------------
Thanks for creating a pull request to contribute to the KiCad libraries! To speed up integration of your PR, please check the following items:

- [ ] Provide a URL to a datasheet for the symbol(s) you are contributing
- [ ] An example screenshot image is very helpful
- [ ] Ensure that the associated footprints match the [official footprint library](https://github.com/kicad/kicad-footprints)
- [ ] If there are matching footprint PRs, provide link(s) as appropriate
- [ ] Check the output of the Travis automated check scripts - fix a [... cut]
commit ec6d9385 (2018-03-18): Remove slash from name of LM334Z-LFT1

Fixes #390.
closed 2018-03-18 by evanshultz