KiCad/kicad-symbols #412: Some operational amplifiers seem to have the power pins on every unit (visible)

poeschlr, 2018-03-23:
According to this forum post https://forum.kicad.info/t/filter-by-number-of-pins-broken/10076/5 this might make problems with the pin number filter. (It is also against what i remember we decided some time back)

jkriege2, 2018-03-23:
yes, I think they haven't been updated yet ... mostly because there was no definite decision (if I remeber correctly) how to handle them with 1-4-unit variants)

evanshultz, 2018-07-13:
Fixed with https://github.com/KiCad/kicad-symbols/pull/578.

However, at https://github.com/KiCad/kicad-symbols/pull/578#issuecomment-390287547 I opened a related discussion:
An option would be to put the triangle symbol on all units. Then it's obvious if many pins are on the power unit as they're not floating due to having some containing shape, and it still allows stacking the power unit on other units.
[image]

poeschlr, 2018-07-27:
What is the state of this? If i remember correctly then this should be fixed now.

evanshultz, 2018-08-24:
It is all done.

closed 2018-08-24 by evanshultz