KiCad/kicad-symbols #418: Spice

arildj78, 2018-03-25:
I\m trying to get started with KiCad and Spice, but it I can't seem to find any parts with spice-models loaded. Right-clicking a part->opening properties->Edit Spice Model seems to be empty on all parts that I'm checking... Are there special libraries for spice or am I looking in the wrong place?

diggit, 2018-04-01:
I guess it is because of model licensing being incompatible with KiCad repositories.

evanshultz, 2018-04-02:
@arildj78 
There is a library named `pspice` but it's marked as legacy. Did you check that library?

Right now, I honestly don't know. Let me take a look at this soon (hopefully with 24 hours). I haven't tried simulation yet in KiCad and at least one of the librarians should be well-versed in it. I am very familiar with circuit simulation from other tools so let me see if I can help.

evanshultz, 2018-04-02:
I figured I should start with Wayne's video at https://fosdem.org/2018/schedule/event/cad_kicad_v5/, since it publicly showed this off recently. Start at around 9:30. He references a video by Tom and Orson, which I believe is at https://www.youtube.com/watch?v=A2_-hdRcf4U.

I do not, at this point, believe any parts in the library have really been intended for circuit simulation. I could be wrong.

SPICE directives placed on the schematic page work OK. I can transform a resistor or other 2-terminal symbol into a seemingly functional V or I source. And simple parts seem to work fine using the R, L, and C primitives. But I currently haven't figured out how to get a diode or other more complex primitive to work. I can load in an external library file and it is parsed correctly, but the model isn't being attached. Only after that would I tackle subcircuits.

Are you also able to work with the normal SPICE primitives like R, L, C and sources? Those seem easiest to get going.

I'll try to figure out more tomorrow. There seem to be a number of analogs to LTSpice in the way things work in general, but I need to figure out the details. I feel pretty confident creating a nicer library of SPICE parts will be a longer-term objective that won't be ready for KiCad v5.

arildj78, 2018-04-04:
I've started to figure this out by now but it wasn't intuitive. I was hoping to see the SPICE directives when browsing the SPICE MODEL and did not realize that changing between the different tabs also changed the behavior. Often this kind of behavior is controlled by radiobuttons and not by tab selection.

arildj78, 2018-04-06:
I just realized that part of the problem was me choosing operating point as the simulation mode. I'm now able to run a successful transient analysis on a simple resistor puzzle, but when I'm changing to operating point i get this:

```
Circuit: KiCad schematic
Doing analysis at TEMP = 27.000000 and TNOM = 27.000000
 Reference value :  0.00000e+000
No. of Data Rows : 1
Background thread stopped with timeout = 0
Error: no such vector 2
Error: no such vector 1
Error: no such vector 3
```

evanshultz, 2018-04-06:
Ah. Yes. Can you attach your schematic? Let me look into it?

BTW, can you run the 3 simulations in the `<KiCad install path>\share\kicad\demos\simulation` as I mentioned above?

arildj78, 2018-04-07:
[SpiceTest.zip](https://github.com/KiCad/kicad-symbols/files/1886362/SpiceTest.zip)

List of issues
-------------------------------
1 .op directive as well as Operating Point tab in simulation settings fail
  with    Error: no such vector 1

2 I can't get D1 to show diode like behaviour during DC sweep. Neither by
  providing the model via a file or adding it to a comment.

3 It would be nice to be able to cut-and-paste a model into the SPICE model
  editor. As it is now, I need to provide a separate file with the model
  data or understand how to include it as a comment.

4 It took ages to understand why I couldn't reference V1 in the command
  ".dc v1 0 9 0.1" If the SPICE command was shown after inserting values in
  the settings dialog it could be educational. The solution was to reference it
  as VV1. If opening the simulation settings dialog with .dc uncommented in
  the schema, it shows the DC Source as VV1. When opening the dropdown menu
  the only option available is V1 - this looks like a bug.

5 The simulation settings dialog box should remember the last settings used.
  As it is now, I need to figure out the correct string and enter it in a text
  box on the schema.

6 It would be nice if the Power library included a SPICE powersource that was
  preconfigured. It took me a while to figure out that the different powersymbols
  e.g. +9V or Vcc wasn't ment to be used as a SPICE powersource.

arildj78, 2018-04-07:
I think I have solved the problem with D1. After rotating it 180 degrees it seems to work with both the supplied .lib file and with model data supplied from a comment in the schematics. From this I'm guessing that the symbol I chose had the pin assignment backwards.

[1N4007.REV0.LIB.txt](https://github.com/KiCad/kicad-symbols/files/1886400/1N4007.REV0.LIB.txt)

arildj78, 2018-04-07:
> BTW, can you run the 3 simulations in the <KiCad install path>\share\kicad\demos\simulation as I mentioned above?

Yes I can. I tested the following projects with the simulation modes .tran, .ac and .op
```
pspice        C:\Program Files\KiCad\share\kicad\demos\pspice\pspice.pro
laser_driver  C:\Program Files\KiCad\share\kicad\demos\simulation\laser_driver\laser_driver.pro
rectifier     C:\Program Files\KiCad\share\kicad\demos\simulation\rectifier\rectifier.pro
sallen_key    C:\Program Files\KiCad\share\kicad\demos\simulation\sallen_key\sallen_key.pro
```

Here are the text output and my comments

# pspice
## .tran 1u 10m
```diff
Circuit: KiCad schematic
Background thread stopped with timeout = 0
Doing analysis at TEMP = 27.000000 and TNOM = 27.000000
Initial Transient Solution
(...)
```
The simulation works flawlessly

## .ac dec 100 10 1Meg
```diff
Circuit: KiCad schematic
Doing analysis at TEMP = 27.000000 and TNOM = 27.000000
 Reference value :  1.00000e+001
No. of Data Rows : 501
```
The simulation works flawlessly

## .op
```diff
Circuit: KiCad schematic
Doing analysis at TEMP = 27.000000 and TNOM = 27.000000
 Reference value :  0.00000e+000
No. of Data Rows : 1
Background thread stopped with timeout = 0
-Error: no such vector 3
-Error: no such vector 6
-Error: no such vector 9
-Error: no such vector 5
-Error: no such vector 8
-Error: no such vector 4
-Error: no such vector 2
-Error: no such vector 13
-Error: no such vector 7
-Error: no such vector 12
-Error: no such vector 11
-Error: no such vector 10
-Error: no such vector 1
```
The simulation does not give any data

# laser_driver
## .tran 10p 150n
```diff
Circuit: KiCad schematic
-Warning: Model issue on line 1 :
.model laser d(is=1e-22 rs=6 n=1.5 cjo=50p xti=100 iave=160m vpk=5) ...
-unrecognized parameter (iave) - ignored
-unrecognized parameter (vpk) - ignored
Reducing trtol to 1 for xspice 'A' devices
Doing analysis at TEMP = 27.000000 and TNOM = 27.000000
-Warning: vv1: no DC value, transient time 0 value used
Initial Transient Solution
(...)
```
The simulation works but it is apparently using a default diode instead of the laserdiode and it seems as if there is missing a DC value but I can't see the effect of this on the result.
[... 94 lines cut ...]
No. of Data Rows : 61
```
The simulation works but it seems as if there is missing a DC value but I can't see the effect of this on the result.

## .op
```diff
Circuit: KiCad schematic
Background thread stopped with timeout = 0
Reducing trtol to 1 for xspice 'A' devices
Doing analysis at TEMP = 27.000000 and TNOM = 27.000000
Warning: vv1: has no value, DC 0 assumed
 Reference value :  0.00000e+000
No. of Data Rows : 1
Background thread stopped with timeout = 0
-Error: no such vector 3
-Error: no such vector 6
-Error: no such vector 2
-Error: no such vector 1
-Error: no such vector 4
-Error: no such vector 5
```
The simulation does not give any data

# Comment on the editing of Symbol Properties
It seems as if i can enter a SPICE source in the components Value field such as `AC 1 SIN(0 3 1k)`. If I then press Edit Spice Model I can further change my values but when I press OK, the values are stored in a new field named Spice_Model. The symbol now has two fields showing data for the SPICE source, both the **Value field** and the **Spice_Model** but only the latter has effect on the simulation. If the Spice_Model field gets deleted, the Value field once again takes effect. This behaviour is confusing especially for unexperienced users such as myself.

evanshultz, 2018-08-08:
@arildj78 
Can you give me an update how this is going? Have you figured out more since the comment above? I've been able to spend some time on SPICE simulation the last couple days and I'd like to help ferret out issues and be able to provide better support for KiCad users who want to simulate.

poeschlr, 2018-08-08:
We have a few users over at the forum who are also able to help out. (holger is the one who is quite active. but also @bobc Just look in the simulation category https://forum.kicad.info/c/schematic/simulation-ngspice)

evanshultz, 2018-08-08:
Note that I'm using the 5.0.0 released version.
> .op directive as well as Operating Point tab in simulation settings fail
> with Error: no such vector 1

Yep, broken for me too.

> 
> I can't get D1 to show diode like behaviour during DC sweep. Neither by
> providing the model via a file or adding it to a comment.

Yes, pin 1 is assumed to be the anode in a SPICE D model. You can add the property "Spice_Node_Sequence" with a value of "2 1" in the UI to address this:
[image]

> It would be nice to be able to cut-and-paste a model into the SPICE model
> editor. As it is now, I need to provide a separate file with the model
> data or understand how to include it as a comment.

This can be done, at least it can be done now. The model name must match the VALUE field of the symbol. This may have changed since you tested, but I removed all properties (leaving only the top two), kept the `.model D1n4007...` text, and set the symbol VALUE to `D1N4007`. Works now.

> It took ages to understand why I couldn't reference V1 in the command
> ".dc v1 0 9 0.1" If the SPICE command was shown after inserting values in
> the settings dialog it could be educational. The [...]

It simulates with `.dc V1 0 9 0.1` now, but the UI still shows `VV1` so it seems there is work to be done:
[image]

> The simulation settings dialog box should remember the last settings used.
> As it is now, I need to figure out the correct string and enter it in a text
> box on the schema.

Did you figure this out? I don't see any effect of the `.plot`, `.print`, or `.probe` SPICE directives.

> It would be nice if the Power library included a SPICE powersource that was
> preconfigured. It took me a while to figure out that the different powersymbols
> e.g. +9V or Vcc wasn't ment to be used as a SPICE powersource.

I will handle this in a forthcoming update to the library. See https://github.com/KiCad/kicad-symbols/issues/805

evanshultz, 2018-08-08:
Regarding the `VV1` thing, that was apparently cached. If I close KiCad and try again everything seems OK, so I believe that issue has been fixed.

evanshultz, 2018-08-08:
I attached an updated version of you simulation to this comment.
[SpiceTest.zip](https://github.com/KiCad/kicad-symbols/files/2271934/SpiceTest.zip)

And now for the included simulation file, where I have comments:

# laser_driver
## .tran
The laser diode is using the SPICE diode model with specific parameters. It should be fine. Some parameters given do not apply in ngspice and so are ignored. This should be fine.

As the source doesn't have a DC value assigned (not needed for a pulse source during the transient analysis but needed when finding the operating point), it starts at zero. This is also fine.

# rectifier
Adding the real model is as simple as putting the following as text on the schematic page:
```
.model 1N4148  D(Is=5.84n N=1.94 Rs=.7017 Ikf=44.17m Xti=3 Eg=1.11 Cjo=.95p
+M=.55 Vj=.75 Fc=.5 Isr=11.07n Nr=2.088 Bv=100 Ibv=100u Tt=11.07n)
```

I'm a bit concerned what `Unable to find definition of model 1n4148 - default assumed` means with regards to the actual model used, though...

## .ac
Yep, there is no AC voltage for an AC analysis. Here, any value works (not just the 1.5V you used) and will be 0dB.

# sallen_key
## .tran
Yes, you will need a time-domain AC source to see anything useful.

# Symbol properties
I can reproduce this. It is odd so I created https://bugs.launchpad.net/kicad/+bug/1786116.

EvansMike, 2018-08-29:
I just had the same pin-reversed situation with 1N4001 diode from the spice lib. This made through to board. De-soldering them all now.

bobc, 2018-08-29:
Unfortunately, (ng)spice uses the opposite convention to KiCad libraries for diodes, and it is not feasible to make them the same. 

Therefore devices in spice.lib should not be used for "real" components, only simulation. "Real" components can be used for simulation if the pin order is specified in the symbol property dialog.

Not sure if there is a way to neatly separate the two, for now user education is the only possibility.

poeschlr, 2018-08-29:
I will fill out the description field of these symbols with a clear warning that they are not intended to be used in conjunction with footprints. Maybe we should use the same pin number as with other symbols and pre fill the pin remapping for spice.

---

The kicad pin numbers are not arability chosen. We use the scheme suggested by IPC.

EvansMike, 2018-08-29:
It seems I used the generic spice diode changing it's value to 1N4001. It was some months ago. Clearly I didn't realise that the pin numbering is different. Some time usually goes by before I layout the board and by that time a diode is a diode. 
Thanks for the replies. It's another day of learning.

nhatkhai, 2018-08-30:
I think, if there is a field for pspice pin order map it may be handy for real component where it's spice model pin order is not the same as 1 2 3 ... ?

evanshultz, 2018-08-31:
@nhatkhai 
Yes, there is. See comments above for details.