KiCad/kicad-symbols #429: Add isolation lines to ACDC converter symbols

antoniovazquezblanco, 2018-03-29:
As someone pointed out in https://github.com/KiCad/kicad-symbols/pull/415 it is desirable to clarify that some converters are isolated by adding lines to the symbol.

antoniovazquezblanco, 2018-03-29:
Related issue https://github.com/KiCad/kicad-symbols/pull/427

Ratfink, 2018-03-30:
Could someone provide a reference document that talks about these lines in schematic symbols?  I'm not familiar with them, but that could just be my I'm-not-an-EE showing.

poeschlr, 2018-03-30:
I always thought the diagonal double line was standardized but i can't find a reference to it. (So i might be wrong)

The only things i could find are these two images as reference for what we could use. I have a slight preference for the one using the diagonal line. But the other one might make fitting pin names easier.

[image]
[image]

[image]

evanshultz, 2018-03-30:
I'm not familiar with any standard here. It's an important thing to communicate, but it's done differently by various folks who provide isolated products.

I skimmed through a few UL and IEC standards and didn't see any graphics they provide. I assume that would be considered a standard we could leverage if they had anything.

IMO, dashed vertical lines are nicer for building symbols as mentioned above. They also communicate something different than a solid line (usually a wire) in a black & white image (like a printout). It's unfortunate KiCad added line styles in Eeschema for v5 but not in the symbol editor. :(

More examples of integrated power supplies:
https://www.cui.com/product/resource/pbk-1.pdf
[image]

http://cdn.vicorpower.com/documents/datasheets/vibrick/VI_BRICK_175V_48V_330W_PFM.pdf
[image]

http://www.meanwell.com/Upload/PDF/IRM-02/IRM-02-SPEC.PDF
[image]

https://media.digikey.com/pdf/Data%20Sheets/Telcodium%20PDFs/TD-PFC-500_Series_Ds.pdf

http://www.tamura-ss.co.jp/electronics/pdf/JP/MPM/MPM1517SJ_1612BE.pdf
[image]

http://www.st.com/content/ccc/resource/technical/document/datasheet/a8/01/14/4b/55/24/4b/3c/CD00173482.pdf/files/CD00173482.pdf/jcr:content/translations/en.CD00173482.pdf
[image]

And some ICs:
https://www.silabs.com/documents/public/data-sheets/Si824x.pdf
[image]

http://www.ti.com/lit/ds/symlink/iso3082.pdf
[image]

http://www.ti.com/lit/ds/symlink/iso7240c.pdf
[image]

http://www.ti.com/lit/ds/symlink/isow7841.pdf
[image]

https://www.silabs.com/documents/public/data-sheets/Si8751-2.pdf
[image]

https://www.silabs.com/documents/public/data-sheets/Si838x-DataSheet.pdf
[image]

http://www.analog.com/media/en/technical-documentation/data-sheets/ADuM4120-4120-1.pdf
[image]

http://www.analog.com/media/en/technical-documentation/data-sheets/ADuM6410-6411-6412.pdf
[image]

And transformers:
http://catalog.triadmagnetics.com/Asset/N-55MG.pdf
[image]

https://www.hammfg.com/files/parts/pdf/171E.pdf
[image]

@Ratfink 
The lines show there is no electrically-conductive connection between input and output. Often the isolation will be done galvanically, optically, capacitively, etc. This is very common in AC-DC power supplies which accept AC mains voltage as an input, using a transformer for galvanic isolation, because it is by far the most common way to span the primary-secondary barrier and meet worldwide safety regulations while providing "touchable" secondary voltage(s).

Ratfink, 2018-03-30:
@evanshultz Yes, I understand what it's showing, I'm just not familiar with that particular way of showing galvanic isolation.  I was wondering if the double-dashed-line is a standard I'm not familiar with or if it was chosen arbitrarily.

I think that dashed vertical lines are a reasonable way of showing galvanic isolation, assuming no standard can be found to suggest anything different.  I would like to see them centered better than in the screenshots in #415, though.

poeschlr, 2018-03-31:
The problem with dashed lines is that kicad has no support for them. Which means the contributor would need to draw many short lines by hand.

antoniovazquezblanco, 2018-04-18:
Any updates/decisions on the matter?

I think having isolation information in the symbol is important regardless of what drawing style you decide to use.

Oftopic: There seems to be a GH plugin to make polls in issues so that the involved people can vote.

poeschlr, 2018-04-18:
As there is no real standard we are completely free to choose whatever we want.
Or in other words: The first contributor is free to choose.

---

If you press me i will say that i kind of like the diagonal line with a second shorter diagonal line as shown in the second image in my post. (Not sure if this works with pin names inside. If not maybe we can use the same thing by using vertical lines.) 

I would not use dashed lines as they are not supported in kicad and only create more work for contributors.

antoniovazquezblanco, 2018-04-18:
I would agree that this is the one I like the most:

[image]

Maybe if the symbol is not square it will look odd but there is no harm in trying. I would try to keep the lines 45º even when the body ratio is not 1:1 but up to the contributor :).

Thanks!

Ratfink, 2018-04-18:
I'm not too crazy about the 45° diagonal line.  It wouldn't work well with square symbols with named pins, e.g. in the example above, if the pins at the top right and bottom left had names, they'd need to be moved.

A vertical line (or lines) works better in that regard, doesn't create questions about symbol geometry, and is evocative of a transformer, perhaps the most classic example of galvanic isolation.  It doesn't have to be dashed, it could be a double or triple solid line like in some of the examples above.  I really like the look of this:
[image]
So I drew an example of what it would look like in KiCad:
[image]
The gaps could be smaller if people prefer (I think I prefer this one, with 25 mil between the edge of each rectangle and the center line):
[image]

poeschlr, 2018-04-18:
I first also thought this would work but to be honest it looks to me like two separate objects.

A way around it is by putting the pin names outside. And yes i am sure there will be parts that will create problems with either solution we will come up with.
[image]

lorem-ipsum, 2018-04-18:
The diagonal line has the problem that it will be difficult to implement for more complex symbols. There are for example the ADuM5000 USB isolators. Placing everything outside would look weird on those. And at least in my opinion the symbols for galvanic isolation should be consistent throughout the libraries.
Looking at the suggestions I prefer the second variant of @Ratfink with smaller distances.

Ratfink, 2018-04-18:
Good point about the USB isolators.  Symbols that have pins on the top and bottom on both sides are incompatible with a diagonal isolation line.

Looking through the library more, I see symbols like ADuM1412, which have graphic elements spanning the isolation lines.  These aren't compatible with my suggestion above, but they work fine with the style they have now (two vertical dashed lines in a single body rectangle).  To have one consistent style, I'd now suggest we stick with a style like ADuM1412 (two vertical dashed lines) which is already prevalent in the library, but maybe set a standardized line spacing and style as there's currently quite a bit of variation in Isolator.lib.

evanshultz, 2018-04-18:
Humbug to diagonal lines.

I infer from Wayne's talk at FOSDEM 2018 that line styles will be coming to the symbol editor soon (they arrive in Eeschema for v5) so hopefully it won't be such a pain in the near future. But we still torture ourselves in the interim. Perhaps solid vertical lines are required for now, with dashed lines an option, and then make all dashed (if dashed lines win the vote) once/if line styles arrive for symbols?

antoniovazquezblanco, 2018-05-23:
To move fordward with the issue;
1. Do the librarians agree that isolation lines should be used to indicate galvanic isolation?
2. If so, do they agree that vertical dashed lines when possible is a good solution as a way to indicate this?

If both are afirmative I may open an issue to add the task to document this in the KLC at https://github.com/KiCad/kicad-website/issues and I will try to compile a list of symbols that are missing this feature if this is ok with everyone.

jkriege2, 2018-05-23:
I would agree to both!

evanshultz, 2018-05-25:
I agree. Thanks! And as I mentioned above, you didn't specify line length (only line width and separation).

antoniovazquezblanco, 2018-11-08:
Converter_ACDC recap. The following symbols should be unified.

Symbols without vertical lines:
- HS40003
- IRM-02-3.3
- IRM-02-3.3S
- TMLM10103
- VTX-214-015-103

Symbols with one vertical line:
- IRM-03-3.3S
- TMLM04103
- TMLM04225
- TMLM05103

Symbols with two vertical lines:
- IRM-20-3.3

Waiting for input on which style should be used.

antoniovazquezblanco, 2018-11-13:
From https://github.com/KiCad/kicad-website/issues/284#issuecomment-437988261

The way fordward is single line:
[image]

antoniovazquezblanco, 2018-11-13:
Fix submitted

closed 2018-11-20 by evanshultz