KiCad/kicad-symbols #441: Vendor-specific diode symbols?

evanshultz, 2018-03-30:
We have the following symbols in Diode with manufacturer names:
Central_Semi_CMKD4448
Central_Semi_CMKD6001
Comchip_ACDSV6-4448TI-G
Comchip_CDSV6-4148-G
Comchip_CDSV6-4448TI-G

Even if only one vendor makes a part, we don't normally do this. Why here?

If there's no reason, should this be "fixed"?

@SchrodingersGat @poeschlr @jkriege2 
You guys have been around a while. Any ideas?

jkriege2, 2018-03-31:
I think it should be fixed.

poeschlr, 2018-03-31:
I would remove the manufacturer prefix from the name. Unless there is a name conflict in the same lib.

evanshultz, 2018-03-31:
Alright. I plan to take a look after https://github.com/KiCad/kicad-symbols/pull/442 is merged.

poeschlr, 2018-08-30:
I fear fixing this now has to wait till we start the v6 development. New symbols however should follow the rule laid out here.