KiCad/kicad-symbols #449: RFM9x pin name

sdalu, 2018-04-02:
Should certainly use ~RESET (for pin 6) and ~SS (for pin 5).
  (only checked for the RFM95)

poeschlr, 2018-04-02:
Always link a datasheet that supports you claim. It might also be a good idea to add a screenshot of the symbol that is currently in the lib (Make sure you use the current HEAD of the master branch)

sdalu, 2018-04-02:
http://www.hoperf.com/upload/rf/RFM95_96_97_98W.pdf.
* Section 7.2.2 for RESET (called NRESET). 
* section 4.3 for NSS

Checking in HEAD, file as moved to RF_AM_FM, but text show its still; RESET and NSS
Don't know how to used HEAD in my kicad install

evanshultz, 2018-04-02:
@sdalu 
By HEAD we mean the current commit (and not an old version of the library).

Secondly, you should give the library you're talking about. In this case it is `RF_AM_FM.lib`, and was mentioned in your second post, but it will save time from someone searching for the symbol.

Here are the relevant datasheets:
- [RFM69HW](http://www.hoperf.com/upload/rf/RFM69HW-V1.3.pdf)
- [RFM95-98](http://www.hoperf.com/upload/rf/RFM95_96_97_98W.pdf) [as you linked above]

Interestingly, I do not see RFM97W at http://www.hoperf.com/rf_transceiver/, nor does it appear in a search on the site. Perhaps it's obsolete?

Now that enough information is here to figure out to what you refer, let's look into it.

NSS: Both datasheets use the terminology `NSS` for the SPI reset pin. It appears to be a standard SPI chip select, pulled high and driven low to talk to the module. **I would not object to renaming this pin to `~NSS`, but I do not agree with changing it to `SS`.**
RFM69: pages 10 & 44; RFM9x: pages 11 & 75

RESET: This is more complicated. This pin is listed at I/O on both modules which I found weird at first. But  And the section describing power on reset shows an internal POR will drive RESET at some time after VDD is applied when the module is ready to use. This is the Output function of RESET because the user will need to read the RESET pin being internally driven and then wait before commencing SPI data transfer. Oddly, the RESET pin seems to be driven internally to a different logic state after power on (RFM69 is shown driven low while RFM9x is shown driven high). The Manual Reset section of the datasheet shows the pin being driven into reset in opposite logic levels as well (low for RFM9x, as you said, but high for RFM69). **I would not object to renaming this pin to `~RESET` for RFM9x, but not for RFM69. Also this pin is Bidirectional in the RFM9x symbol but only at Input on the RFM69 symbol.**
RFM69: pages 10 & 75-76; RFM9x: pages 11 & 108-109

While a relatively minor thing, I think you did identify a place where our libraries could be improved. However, I suggest a slightly different resolution above. Would you be interested in submitting a fix?

sdalu, 2018-04-02:
I'm just an occasional KiCad user (without much electronic knowledge), took a little time to share something that seemed strange to me, but I would prefer a person more skilled than me submit a PR

sdalu, 2018-04-02:
Seemed to be that the N was for the "negation", and kicad used ~ for that. 
SS is for Slave Select (according to sparkfun) (sometime also called CS), that was why the ~SS suggestion

Also pin 6 is called RESET in the hoperf documentation, but called NRESET in the SX1276[1]

It doesn't look like the use of SS/CS ~SS/~CS is consistant across the kicad libraryies Don't know if there is a rule for that.

[1] https://www.semtech.com/uploads/documents/DS_SX1276-7-8-9_W_APP_V5.pdf

closed 2018-05-19 by sdalu