KiCad/kicad-symbols #451: RC2 is near. Is there anything big that anybody wants to be done before that?

poeschlr, 2018-04-03:
Mailing list announcement: https://lists.launchpad.net/kicad-developers/msg35171.html

I think we might want to introduce something similar to a feature freeze as soon as RC2 is tagged. (So no library or part renaming. No major changes to existing parts -> examples of things that might be a bad idea during that time frame: symbol size changes, changing pin placements, changing the number of units, ....)

So if you have something planned that you really want to be included in version 5, you might need to do it now.

If there is a known major problem that we can not fix till the end of this week we might want to talk to the developers. (It might require the release date to be pushed back.)

poeschlr, 2018-04-03:
Forgot to tag you guys: @evanshultz, @jkriege2, @Misca1234, @Ratfink, @Shackmeister

poeschlr, 2018-04-03:
One can turn off automatic text placement. (In v4 this was in the general preferences dialog. Not sure if this is still the case in v5.) There is nothing we can do about it on the library side so it is not really worth worrying about it. (You can file a bug report or add your voice to this one: https://bugs.launchpad.net/kicad/+bug/1682496)

---

About the connector lib. I totally missed your pull request. I am not sure i will have time this week to review it. (I will try my best.) But it will definitely help if you could fix the merge conflicts. 

For the future please try to split up pull requests such that you have one per topic. (This helps with reviewing. Yes i know for symbol libs it might require more merging / rebasing.)

Ratfink, 2018-04-03:
I'd still like to see #305 make it into 5.0, but I realize that it might be a bit much at this point unless we want to review it with only a few spot checks.

Shackmeister, 2018-04-03:
My current ongoing projects I would like to finish for V5:
New 3D models for LEDS https://github.com/KiCad/kicad-packages3D/pull/280
New SMD Electrolytic Capacitor footprints https://github.com/KiCad/kicad-footprints/pull/372
Fix SMD Electrolytic Capacitor 3D models, the old are rotated 180 degrees
Renamed picoblade 3D models to include the mounting-pins renaming

My wishlist:
Add JST GH connector models 
Go through all the common packages and make sure the 3D models are up to date

evanshultz, 2018-04-03:
@poeschlr 
I unbroke my branch so it can be merged. If anybody has time to review please let me know. I do understand there is a lot covered in that PR.

evanshultz, 2018-04-04:
It isn't big, but I think https://github.com/KiCad/kicad-symbols/pull/454 needs to be merged.

evanshultz, 2018-04-04:
One more potential library change: the library for simulation symbols is named `pspice.lib`. (Symbols in other libraries can be used, too, but there are specification simulation symbols.) PSpice is a specific implementation of the SPICE circuit simulator program and owned by a commercial company. Would it be better to rename to `spice.lib`? Or does someone have another suggestion?

evanshultz, 2018-04-12:
I am still waiting for https://github.com/KiCad/kicad-symbols/pull/310 to get resolved.

And note the simulation library question I posed above.

Finally, something else I noticed today: It seems the major ADC and DAC libraries are called `Analog_ADC` and `Analog_DAC` (there are also audio-specific ADCs, DACs, and CODECs in `Audio`). Why do they have the `Analog_` prefix? Isn't just `ADC` and `DAC` sufficient? An ADC or DAC has both analog and digital sections by definition, so that just seems weird to me.

Ratfink, 2018-04-12:
Putting both ADCs and DACs under then Analog_ prefix does seem to take an awfully digital-centric view of the world.  I'd be fine with renaming them to be less biased. 🙂

jkriege2, 2018-04-13:
We could either rename them to `MixedSignal_ADC/DAC`, or simply `ADC`and `DAC` ... although in my opinion, the first would be better, as then we have a category to sollect more mixed signal devices ...

cu
JAN

evanshultz, 2018-04-23:
What other mixed signal devices could be collected? Isn't it implied that ADCs and DACs are mixed signal and so adding that prefix is superfluous?

evanshultz, 2018-05-01:
Does anybody know where we are with the v5 release?

@poeschlr @jkriege2 @Ratfink @Misca1234 
I found something that I think is imperative to fix before v5. Here is a schematic section of a LM393 ([datasheet](http://www.ti.com/lit/ds/symlink/lm393-n.pdf) and line 63 of [Comparator.lib](https://github.com/KiCad/kicad-symbols/blob/master/Comparator.lib)):
[image]

You see that power pins are placed on both units since they are, in fact, set to be on all units (the third-to-last element for pins 4 and 8 are zero).

The resulting CvPcb interface (filtered by pin count and library):
[image]

As you can see, only 10 pin packages are listed. The packages for this part have 8 pins! Most (all?) multi-unit opamps and comparators may be affected.

I wrote this issue up at https://github.com/KiCad/kicad-symbols/issues/164, but didn't realize how serious it was until just now.

Unless someone objects, I'll start on this and hope to finish by the end of the week. Do we have a tighter time constraint? This means please do not merge any PRs into Comparator.lib or Amplifier*.lib or Driver*.lib or Audio.lib. Are there other libs likely to be affected?

On a smaller scale, I noticed Package_DIP has several packages missing a pin and named `DIP-5-6...` which doesn't match our new paradigm. I didn't look closely, but I believe they should be `DIP-6-N5...`.

evanshultz, 2018-05-01:
I forgot to include how to fix it and ask one question about the fix: how should the power gate look?

First, here is the symbol with a separate power gate and not interchangeable units:
```
# LM2903_
#
DEF LM2903_ U 0 20 Y Y 3 L N
F0 "U" 150 150 50 H V C CNN
F1 "LM2903_" 250 -150 50 H V C CNN
F2 "" 0 0 50 H I C CNN
F3 "" 0 0 50 H I C CNN
$FPLIST
 SOIC*3.9x4.9mm*P1.27mm*
 DIP*W7.62mm*
 MSOP*3x3mm*P0.65mm*
 VSSOP*2.3x2mm*P0.5mm*
$ENDFPLIST
DRAW
P 4 0 1 10 -200 200 200 0 -200 -200 -200 200 f
X ~ 1 300 0 100 L 50 50 1 1 C
X _ 2 -300 -100 100 R 50 50 1 1 I
X + 3 -300 100 100 R 50 50 1 1 I
X + 5 -300 100 100 R 50 50 2 1 I
X _ 6 -300 -100 100 R 50 50 2 1 I
X ~ 7 300 0 100 L 50 50 2 1 C
X V- 4 -100 -300 150 U 50 50 3 1 W
X V+ 8 -100 300 150 D 50 50 3 1 W
ENDDRAW
ENDDEF
```

Images:
[image]
[image]

And the question: I made the power gate a triangle because that keeps a single shape for all symbols. However, if we want to be able to stack the power pins without any weird graphical artifacts we will need power pins without a containing box. This also works, but doing it this way creates double refdes/value which is also weird:
[image]

```
# LM2903__
#
DEF LM2903__ U 0 20 Y Y 3 L N
F0 "U" 150 150 50 H V C CNN
F1 "LM2903__" 250 -150 50 H V C CNN
F2 "" 0 0 50 H I C CNN
F3 "" 0 0 50 H I C CNN
$FPLIST
 SOIC*3.9x4.9mm*P1.27mm*
 DIP*W7.62mm*
 MSOP*3x3mm*P0.65mm*
 VSSOP*2.3x2mm*P0.5mm*
$ENDFPLIST
DRAW
P 4 1 1 10 -200 200 200 0 -200 -200 -200 200 f
P 4 2 1 10 -200 200 200 0 -200 -200 -200 200 f
X ~ 1 300 0 100 L 50 50 1 1 C
X _ 2 -300 -100 100 R 50 50 1 1 I
X + 3 -300 100 100 R 50 50 1 1 I
X + 5 -300 100 100 R 50 50 2 1 I
X _ 6 -300 -100 100 R 50 50 2 1 I
X ~ 7 300 0 100 L 50 50 2 1 C
X V- 4 -100 -300 150 U 50 50 3 1 W
X V+ 8 -100 300 150 D 50 50 3 1 W
ENDDRAW
ENDDEF
```

poeschlr, 2018-05-01:
I think there are still some things to do before the rc2 release (or wayne never officially created it)

And after rc2 there will be some time for the documenters and i assume wayne will ask if everyone is ready before the release.

evanshultz, 2018-05-01:
OK. Good. Let me know if this isn't a serious problem, but I'm assuming it is and needs to be addressed.

How should this be fixed:
1. Power pins on another triangle; this means a big symbol for the power pins that is mostly blank.
2. Power pins floating; this allow laying over another gate but there are pairs of refdes/value.
3. Something else?

I mentioned another issue above. It's not quite the same problem, but while I'm in there I might as well separate the symbols by package.

poeschlr, 2018-05-01:
The problem is quite serious to be honest. If the user connects the pins to two different nets, some weird stuff can happen. (If i remember correctly there where a few topics on the forum where we discovered that. As it has been quite long ago i can't really remember enough details for me being able to begin to search for it.)

I think we decided somewhere for the power pins being placed on a separate unit and floating. (only if the symbol is already a multi unit symbol.)
If i remember correctly it is a topic by jan. (Githubs search functionality is really lacking or i don't understand how to use it.)

jkriege2, 2018-05-02:
I would vote for floating pins ... the issue where we discussed such stuff already is here: https://github.com/KiCad/kicad-symbols/issues/190

stambaughw, 2018-05-02:
Rene,

I'm still waiting on two critical bug fixes before rc2 can be released.
As soon as they are addressed, I will tag rc2 and make a release
announcement.

Cheers,

Wayne

poeschlr, 2018-05-02:
@stambaughw thanks for the update.

evanshultz, 2018-05-07:
The PR to fix duplicate power pins multi-unit symbols is at https://github.com/KiCad/kicad-symbols/pull/578.

BTW, in reference to my own question about which symbols could go into a `MixedSignal` library at https://github.com/KiCad/kicad-symbols/issues/451#issuecomment-383674800, we currently have `AD5593R` in `Analog` and it could go there. It's a combo ADC or DAC or GPIO (you pick which one) for 8 pins in the same package.

poeschlr, 2018-05-18:
There is an update on the mailing list. RC2 will be tagged this Sunday. More details see: https://lists.launchpad.net/kicad-developers/msg35785.html

poeschlr, 2018-05-20:
Ok i think most of the things here are done. Nothing that is still open is important enough to postpone rc2 so i tag the libs now.
@evanshultz, @Ratfink, @jkriege2, @Shackmeister we should no longer rename libs until after the v5 release. (Meaning the current lib names are fixed for the v5 release cycle. Adding additional libs is not a problem it just means people would not necessarily get these additional libs as they might not notice that they need to update their lib tables.)

closed 2018-05-20 by poeschlr