KiCad/kicad-symbols #458: Where is maxim.dcm?

joe-skb7, 2018-04-06:
Hi,

I can see `maxim.dcm` here: https://github.com/KiCad/kicad-library/tree/master/library
But in this new repo it seems to be missing. Can I ask why, and what can be done about it? Basically I contributed the symbol, footprint and 3D model for MAX7325 some time ago, and I'd like to have it in official libraries. Please advice.

Thanks!

joe-skb7, 2018-04-06:
Seems logical. Thank you for the info, Jan! I'm glad my stuff is in the lib, gonna use it soon in open-source project.

closed 2018-04-06 by joe-skb7

poeschlr, 2018-04-06:
As a tip for easily finding a symbol in the lib:

Make a dummy project where you add all symbol libs. Use eeschemas "add component" dialog as a library search.