KiCad/kicad-symbols #464: Where to put the SPICE model?

arildj78, 2018-04-08:
I have just forked kicad-symbols and are in the process of adding the AD795 OpAmp to the library. In addition to creating the symbol and footprint, I would like to include the SPICE model from Analog Devices. How should this be done? The only way I've found so far is to give the symbol the path to a text file with the SPICE subcircuit, but I expect this link to be broken when the library is used on any computer but mine.

tijuca, 2018-04-08:
Side note, please check if the license for the SPICE model is covered by the license for the git tree. Especially for Debian I need to keep track of not DFSG compatible parts. So far I can see Analog Devices uses a non-free license.

http://www.analog.com/en/license/spice-models?mediaPath=media/en/simulation-models/spice-models/ad795.cir&modelType=spice-models

> License agreement for Spice Models
> LICENSE AGREEMENT
> 
> IMPORTANT NOTICE. Read this Agreement carefully before downloading or using this spice model. BY DOWNLOADING OR USING THIS SPICE MODEL IN ANY WAY YOU ACKNOWLEDGE THAT YOU HAVE READ, UNDERSTAND AND AGREE TO THE TERMS OF THIS AGREEMENT. IF YOU DO NOT AGREE TO THESE TERMS OR IF YOU DO NOT HAVE THE AUTHORITY DESCRIBED BELOW, DO NOT DOWNLOAD THIS SPICE MODEL, DO NOT USE THIS SPICE MODEL IN ANY WAY, AND PROMPTLY DELETE OR DESTROY ANY COPIES OF THIS SPICE MODEL IN YOUR POSSESSION
> 
> If You are entering into this Agreement on behalf of a company or other organization, You represent that You have the authority to bind it to this Agreement and commit funds on its behalf, and the terms "You" and "Your" will refer to that company or organization.
> 
> Analog Devices SPICE Model Library
> 
> Copyright 1995-2015 by Analog Devices, Inc.
> 
> The information in this SPICE Model is protected under United States copyright laws. ADI hereby grants You a nonexclusive, nontransferable license to use this SPICE Model as long as You abide by the terms of this Agreement.
> 
> You may not sell, loan, rent, lease, or license the SPICE Model, in whole, in part, or in modified form, to anyone outside Your company. You may modify this SPICE Model to suit Your specific applications, and You may make copies of this SPICE Model for use within Your company.
> 
> You may include copies of Analog Devices’ SPICE models with any software you sell or distribute. However, you may not make changes to the redistributed copies of Analog Devices SPICE models other than to:
> 
>
> 1. Include comments.
>  2. Change nomenclature so that it will run on Your company’s software. No changes may be made that affect the performance or function of the model.
> 
> Analog Devices includes SPICE models in its library that have been developed by third parties. These models may not be redistributed.
> 
> This SPICE Model is provided AS IS, WHERE IS, AND WITH NO WARRANTY OF ANY KIND EITHER EXPRESSED OR IMPLIED, INCLUDING BUT NOT LIMITED TO ANY IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE.
> 
> ADI'S LIABILITY FOR DAMAGES TO YOU FOR ANY CAUSE WHATSOEVER, REGARDLESS OF THE FORM OF ANY CLAIM OR ACTION, SHALL NOT EXCEED ONE HUNDRED US DOLLARS ($100.00 US). IN NO EVENT SHALL ADI BE LIABLE FOR ANY LOSS OF DATA, PROFITS OR USE OF THE SPICE MODEL, OR FOR ANY SPECIAL, INCIDENTAL, CONSEQUENTIAL OR OTHER INDIRECT DAMAGES ARISING OUT OF OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THE SPICE MODEL.
> 
> This Agreement shall be governed by and construed in accordance with the laws of the Commonwealth of Massachusetts, without regard to its choice of law provisions. The parties agree that the United Nations Convention on Contracts for the International Sale of Goods is hereby excluded in its entirety from this Agreement. The sole jurisdiction and venue for all actions related to the subject matter hereof shall be the state and federal courts located in Suffolk County, Massachusetts.

arildj78, 2018-04-08:
I read through that and was not sure how to interpret the License Agreement. Hopefully the quote below could be what is needed for it to be included or maybe Analog Devices need to give permission.

> You may include copies of Analog Devices’ SPICE models with any software you sell or distribute. However, you may not make changes to the redistributed copies of Analog Devices SPICE models other than to:

arildj78, 2018-04-08:
> Anyhow, it's definitely not CC-BY-SA-4.0 (with or without exception).
>
> Sentences like "you may not make changes" are always a sign for a non-free
> license. As SPICE models just plain text I really don't know what companies
> want to protect here ...

Apart from the license problem, how should such a model be placed in the
library?

Here are som text from the KLC

S6.2 - Component fields must be filled appropriately
<http://kicad-pcb.org/libraries/klc/S6.2/>

   1.

   *Reference* field is selected appropriately for the symbol and is
   *visible*
   2.

   *Value* field contains the name of the symbol and is *visible*
   3.

   *Footprint* field contains footprint link for atomic parts, and is empty
   for generic parts. It must be *invisible*
   4.

   *Datasheet* field is left empty, and is *invisible*
   5.

   The symbol contains no other custom fields

Number 5 on the list seems to prohibit the inclusion of SPICE directives in
the library.

evanshultz, 2018-08-08:
I do not believe we will include references to SPICE models for symbols in the official KiCad library. If a user wants to add SPICE models and a reference after symbol placement or in their own library they are free to do so.

If the SPICE model file (*.lib) is placed in the root of the project directory (with the KiCad *.pro file), you do not need an absolute path to the SPICE model. That should answer one item above. Like this:
[image]

@poeschlr 
Correct that we should not reference SPICE models in official KiCad library symbols (at least at this time)?

antoniovazquezblanco, 2018-11-15:
For the time being the library will not be able to include SPICE model contributions. Many pending issues will need to be solved before we can even talk about setting up the infrastructure to accept this kind of contributions: Licensing, how to review the models, where to store them...

Also, as @evanshultz has explained, the user is free to reference any model they wish in their own library but maintaining a set of fields for this is also a big workload that will not be undertaken at the moment.

I hope this solves your original question @arildj78. Do you have any further questions or can we close the current issue?

Thank you!

arildj78, 2018-11-15:
My issue solved. Thanks for the feedback.

Arild

tor. 15. nov. 2018 kl. 11:42 skrev Antonio Vazquez <notifications@github.com

closed 2018-11-15 by antoniovazquezblanco